jneilliii/OctoPrint-BedLevelVisualizer #92: Support for klipper firmware

FixFuture, 2018-10-16:
Hi!
It would be grat if Bed Level Visualizer would support Klipper firmware too.

Klipper firmware mesh works via BED_MESH_MAP -gcode, which prints out mesh map as a json file.

[https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md](https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md)

jneilliii, 2018-10-16:
Looking at that documentation it appears that BED_MESH_OUTPUT may be a more appropriate option as the other BED_MESH_MAP will clear out the currently loaded mesh if I'm reading it correctly. Since I don't have a Klipper installation could you please copy/paste the output of those two commands here for further evaluation?  Please make sure to highlight what you paste in and press the code block button 
[image] in the editor.

FixFuture, 2018-10-17:
Hi! 

**Here is output for BED_MESH_CALIBRATE (prints each result line by line when it's measured) **
`Send: BED_MESH_CALIBRATE
[...]
Recv: // probe at 78.000,31.000 is z=2.410000
Recv: // probe at 183.500,31.000 is z=2.355000
Recv: // probe at 289.000,31.000 is z=2.327500
Recv: // probe at 394.500,31.000 is z=2.292500
Recv: // probe at 500.000,31.000 is z=2.347500
Recv: // probe at 500.000,145.500 is z=2.352500
Recv: // probe at 394.500,145.500 is z=2.332500
Recv: // probe at 289.000,145.500 is z=2.350000
Recv: // probe at 183.500,145.500 is z=2.377500
Recv: // probe at 78.000,145.500 is z=2.435000
Recv: // probe at 78.000,260.000 is z=2.432500
Recv: // probe at 183.500,260.000 is z=2.350000
Recv: // probe at 289.000,260.000 is z=2.317500
Recv: // probe at 394.500,260.000 is z=2.290000
Recv: // probe at 500.000,260.000 is z=2.317500
Recv: // probe at 500.000,374.500 is z=2.257500
Recv: // probe at 394.500,374.500 is z=2.235000
Recv: // probe at 289.000,374.500 is z=2.277500
Recv: // probe at 183.500,374.500 is z=2.285000
Recv: // probe at 78.000,374.500 is z=2.355000
Recv: // probe at 78.000,489.000 is z=2.362500
Recv: // probe at 183.500,489.000 is z=2.310000
Recv: // probe at 289.000,489.000 is z=2.307500
Recv: // probe at 394.500,489.000 is z=2.257500
Recv: // probe at 500.000,489.000 is z=2.255000
Recv: // Mesh Bed Leveling Complete
Recv: // Bed Mesh state has been saved to profile [default]
Recv: // for the current session.  The SAVE_CONFIG command will
Recv: // update the printer config file and restart the printer.
Recv: ok`

**Here is output for BED_MESH_MAP (prints each result line by line when it's measured and mesh_map_output when probing is done)**
`Send: BED_MESH_MAP
[...]
Recv: // probe at 78.000,31.000 is z=2.400000
Recv: // probe at 183.500,31.000 is z=2.352500
Recv: // probe at 289.000,31.000 is z=2.325000
Recv: // probe at 394.500,31.000 is z=2.290000
Recv: // probe at 500.000,31.000 is z=2.337500
Recv: // probe at 500.000,145.500 is z=2.335000
Recv: // probe at 394.500,145.500 is z=2.330000
Recv: // probe at 289.000,145.500 is z=2.347500
Recv: // probe at 183.500,145.500 is z=2.385000
Recv: // probe at 78.000,145.500 is z=2.435000
Recv: // probe at 78.000,260.000 is z=2.432500
Recv: // probe at 183.500,260.000 is z=2.342500
Recv: // probe at 289.000,260.000 is z=2.310000
Recv: // probe at 394.500,260.000 is z=2.290000
Recv: // probe at 500.000,260.000 is z=2.317500
Recv: // probe at 500.000,374.500 is z=2.262500
Recv: // probe at 394.500,374.500 is z=2.235000
Recv: // probe at 289.000,374.500 is z=2.277500
Recv: // probe at 183.500,374.500 is z=2.282500
Recv: // probe at 78.000,374.500 is z=2.352500
Recv: // probe at 78.000,489.000 is z=2.362500
Recv: // probe at 183.500,489.000 is z=2.310000
Recv: // probe at 289.000,489.000 is z=2.300000
Recv: // probe at 394.500,489.000 is z=2.247500
Recv: // probe at 500.000,489.000 is z=2.237500
Recv: mesh_map_output {"max_point": [500.0, 489.0], "z_positions": [[0.054999999999999716, 0.00749999999999984, -0.020000000000000462, -0.05500000000000016, -0.007500000000000284], [0.08999999999999986, 0.03999999999999959, 0.0024999999999999467, -0.015000000000000124, -0.010000000000000231], [0.08749999999999991, -0.002500000000000391, -0.03500000000000014, -0.05500000000000016, -0.027500000000000302], [0.00749999999999984, -0.06250000000000044, -0.06750000000000034, -0.11000000000000032, -0.08250000000000046], [0.017499999999999627, -0.03500000000000014, -0.04500000000000037, -0.09750000000000014, -0.10750000000000037]], "xy_offset": [58.565, -11.0], "min_point": [78.0, 31.0]}
Recv: ok`

**Output for BED_MESH_OUTPUT:**
`Send: BED_MESH_OUTPUT
Recv: // Mesh Leveling Probed Z positions:
Recv: // 0.065000 0.010000 -0.017500 -0.052500 0.002500
Recv: // 0.090000 0.032500 0.005000 -0.012500 0.007500
Recv: // 0.087500 0.005000 -0.027500 -0.055000 -0.027500
Recv: // 0.010000 -0.060000 -0.067500 -0.110000 -0.087500
Recv: // 0.017500 -0.035000 -0.037500 -0.087500 -0.090000
Recv: Mesh X,Y: 13,13
Recv: Search Height: 5
Recv: Interpolation Algorithm: lagrange
Recv: Measured points:
Recv:   0.017500  -0.019496  -0.033416  -0.035000  -0.032613  -0.032243  -0.037500  -0.049619  -0.067459  -0.087500  -0.103848  -0.108230  -0.090000
Recv:   -0.012860  -0.052859  -0.066468  -0.065936  -0.060799  -0.057882  -0.061296  -0.072445  -0.090016  -0.109990  -0.125631  -0.127495  -0.103426
Recv:   -0.010823  -0.053560  -0.069668  -0.071276  -0.067818  -0.066034  -0.069969  -0.080975  -0.097709  -0.116132  -0.129512  -0.128422  -0.100741
Recv:   0.010000  -0.034424  -0.054115  -0.060000  -0.060566  -0.061862  -0.067500  -0.078652  -0.094053  -0.110000  -0.120350  -0.116523  -0.087500
Recv:   0.038560  -0.006008  -0.029048  -0.039671  -0.044945  -0.049894  -0.057500  -0.068701  -0.082391  -0.095422  -0.102602  -0.096696  -0.068426
Recv:   0.066368  0.023395  -0.001842  -0.016430  -0.025858  -0.034019  -0.043210  -0.054131  -0.065884  -0.075977  -0.080320  -0.073224  -0.047407
Recv:   0.087500  0.047757  0.021986  0.005000  -0.007212  -0.017490  -0.027500  -0.037737  -0.047521  -0.055000  -0.057150  -0.049774  -0.027500
Recv:   0.098591  0.063317  0.038783  0.021317  0.008083  -0.002923  -0.012870  -0.022098  -0.030109  -0.035576  -0.036337  -0.029396  -0.010926
Recv:   0.098837  0.068579  0.046757  0.030638  0.018114  0.007703  -0.001451  -0.009578  -0.016283  -0.020545  -0.020717  -0.014526  0.000926
[... cut ...]   -0.008498  -0.012500  -0.012726  -0.006986  0.007500
Recv:   0.076399  0.053547  0.038375  0.027860  0.019717  0.012398  0.005093  -0.002272  -0.009033  -0.013786  -0.014393  -0.007978  0.009074
Recv:   0.064918  0.041588  0.027742  0.019095  0.012448  0.005693  -0.002191  -0.011137  -0.019989  -0.026502  -0.027344  -0.018093  0.006759
Recv:   0.065000  0.035998  0.019733  0.010000  0.002233  -0.006502  -0.017500  -0.030422  -0.043292  -0.052500  -0.052798  -0.037305  0.002500
Recv: 
Recv: ok`

jneilliii, 2018-10-17:
Ok, based on those results the following GCODE command in the plugin settings should work to get the mesh to display.
```
@BEDLEVELVISUALIZER
BED_MESH_OUTPUT
```

FixFuture, 2018-10-25:
Can confirm. This works!

closed 2018-10-25 by FixFuture

shaiss, 2019-03-19:
confirmed this works, ty!

ProfEngr, 2020-02-15:
I can also confirm it works. Only thing I had to do was invert the Y axis on my Ender 3.

technoo10201, 2020-04-03:
> Ok, based on those results the following GCODE command in the plugin settings should work to get the mesh to display.
> 
> ```
> @BEDLEVELVISUALIZER
> BED_MESH_OUTPUT
> ```

Worked for me too ! Thanks a lot ! 💃

larryhan, 2020-04-04:
> Ok, based on those results the following GCODE command in the plugin settings should work to get the mesh to display.
> 
> ```
> @BEDLEVELVISUALIZER
> BED_MESH_OUTPUT
> ```

This does not work for me. Here is my terminal output.

```
Send: BED_MESH_OUTPUT
Recv: // Mesh Leveling Probed Z positions:
Recv: // 0.000000 -0.080000 0.050000 -0.082500 -0.180000
Recv: // -0.017500 -0.090000 -0.127500 -0.127500 -0.230000
Recv: // -0.040000 -0.107500 -0.155000 -0.162500 -0.195000
Recv: // -0.082500 -0.157500 -0.200000 -0.202500 -0.262500
Recv: // -0.160000 -0.247500 -0.285000 -0.287500 -0.317500
Recv: Mesh X,Y: 13,13
Recv: Search Height: 4
Recv: Mesh Average: -0.15
Recv: Mesh Range: min=-0.3201 max=0.0500
Recv: Interpolation Algorithm: lagrange
Recv: Measured points:
Recv:   -0.160000  -0.193693  -0.223261  -0.247500  -0.265792  -0.278107  -0.285000  -0.287613  -0.287675  -0.287500  -0.289990  -0.298632  -0.317500
Recv:   -0.129352  -0.155743  -0.185502  -0.213313  -0.235521  -0.250130  -0.256800  -0.256852  -0.253262  -0.250669  -0.255366  -0.275306  -0.320103
Recv:   -0.103704  -0.127808  -0.156138  -0.183189  -0.205146  -0.219881  -0.226955  -0.227616  -0.224802  -0.223138  -0.228936  -0.250199  -0.296615
Recv:   -0.082500  -0.106327  -0.132716  -0.157500  -0.177809  -0.192068  -0.200000  -0.202623  -0.202253  -0.202500  -0.208272  -0.225772  -0.262500
Recv:   -0.065185  -0.088738  -0.113462  -0.136399  -0.155508  -0.169669  -0.178683  -0.183271  -0.185075  -0.186656  -0.191497  -0.203999  -0.229486
Recv:   -0.051204  -0.073479  -0.097288  -0.119825  -0.139099  -0.153932  -0.163961  -0.169638  -0.172230  -0.173817  -0.177293  -0.186368  -0.205566
Recv:   -0.040000  -0.059990  -0.083786  -0.107500  -0.128292  -0.144372  -0.155000  -0.160484  -0.162181  -0.162500  -0.164897  -0.173879  -0.195000
Recv:   -0.031019  -0.048710  -0.073233  -0.098930  -0.121656  -0.138778  -0.149177  -0.153246  -0.152892  -0.151533  -0.154101  -0.167042  -0.198313
Recv:   -0.023704  -0.041079  -0.066588  -0.093405  -0.116613  -0.133204  -0.142078  -0.144046  -0.141828  -0.140051  -0.145254  -0.165883  -0.212294
Recv:   -0.017500  -0.039537  -0.065494  -0.090000  -0.109444  -0.121975  -0.127500  -0.127685  -0.125957  -0.127500  -0.139259  -0.169938  -0.230000
Recv:   -0.011852  -0.047525  -0.072275  -0.087572  -0.095285  -0.097687  -0.097449  -0.097644  -0.101746  -0.113632  -0.137576  -0.178256  -0.240751
Recv:   -0.006204  -0.069484  -0.089941  -0.084763  -0.068129  -0.051203  -0.042140  -0.046084  -0.065166  -0.098508  -0.142220  -0.189399  -0.230134
Recv:   0.000000  -0.110854  -0.122181  -0.080000  -0.020823  0.028344  0.050000  0.036152  -0.011687  -0.082500  -0.155761  -0.201440  -0.180000

```

jneilliii, 2020-04-04:
Please enable debug logging in settings, restart OctoPrint and then run the Update Mesh button. Then grab the file plugin_bedlevelvisualizer_debug.log from the Logging section of settings and then drag it into a comment here so I can see what it is doing.

larryhan, 2020-04-04:
Thank you for your quick reply.
Attached are my settings and debug log:

Terminal output:
```
Send: G29 T
Recv: // Sending BLTOUCH_DEBUG COMMAND=reset
Recv: // Sending BLTOUCH_DEBUG COMMAND=pin_up
Recv: // probe at 59.000,25.000 is z=1.552500
Recv: // probe at 102.250,25.000 is z=1.467500
Recv: // probe at 145.500,25.000 is z=1.590000
Recv: // probe at 188.750,25.000 is z=1.445000
Recv: // probe at 232.000,25.000 is z=1.382500
Recv: // probe at 232.000,76.250 is z=1.312500
Recv: // probe at 188.750,76.250 is z=1.412500
Recv: // probe at 145.500,76.250 is z=1.415000
Recv: // probe at 102.250,76.250 is z=1.462500
Recv: // probe at 59.000,76.250 is z=1.540000
Recv: // probe at 59.000,127.500 is z=1.507500
Recv: // probe at 102.250,127.500 is z=1.440000
Recv: // probe at 145.500,127.500 is z=1.380000
Recv: // probe at 188.750,127.500 is z=1.380000
Recv: // probe at 232.000,127.500 is z=1.350000
Recv: // probe at 232.000,178.750 is z=1.282500
Recv: // probe at 188.750,178.750 is z=1.347500
Recv: // probe at 145.500,178.750 is z=1.347500
Recv: // probe at 102.250,178.750 is z=1.390000
Recv: // probe at 59.000,178.750 is z=1.467500
Recv: // probe at 59.000,230.000 is z=1.395000
Recv: // probe at 102.250,230.000 is z=1.307500
Recv: // probe at 145.500,230.000 is z=1.262500
Recv: // probe at 188.750,230.000 is z=1.270000
Recv: // probe at 232.000,230.000 is z=1.235000
Recv: // Mesh Bed Leveling Complete
Recv: // Bed Mesh state has been saved to profile [default]
Recv: // for the current session.  The SAVE_CONFIG command will
Recv: // update the printer config file and restart the printer.
Recv: ok
Send: BED_MESH_OUTPUT
Recv: // Mesh Leveling Probed Z positions:
Recv: // 0.002500 -0.082500 0.040000 -0.105000 -0.167500
Recv: // -0.010000 -0.087500 -0.135000 -0.137500 -0.237500
Recv: // -0.042500 -0.110000 -0.170000 -0.170000 -0.200000
Recv: // -0.082500 -0.160000 -0.202500 -0.202500 -0.267500
Recv: // -0.155000 -0.242500 -0.287500 -0.280000 -0.315000
Recv: Mesh X,Y: 13,13
Recv: Search Height: 4
Recv: Mesh Average: -0.15
Recv: Mesh Range: min=-0.3230 max=0.0400
Recv: Interpolation Algorithm: lagrange
Recv: Measured points:
Recv:   -0.155000  -0.183951  -0.214537  -0.242500  -0.264877  -0.280000  -0.287500  -0.288302  -0.284630  -0.280000  -0.279228  -0.288426  -0.315000
Recv:   -0.124136  -0.151924  -0.182820  -0.211286  -0.233556  -0.247643  -0.253333  -0.252191  -0.247554  -0.244537  -0.250031  -0.272703  -0.322994
Recv:   -0.100741  -0.127303  -0.156704  -0.183786  -0.205065  -0.218734  -0.224660  -0.224390  -0.221143  -0.219815  -0.226979  -0.250883  -0.301451
Recv:   -0.082500  -0.107027  -0.134372  -0.160000  -0.180792  -0.195051  -0.202500  -0.204280  -0.202953  -0.202500  -0.208323  -0.227243  -0.267500
Recv:   -0.067562  -0.089040  -0.114657  -0.139866  -0.161406  -0.177299  -0.186852  -0.190657  -0.190591  -0.189815  -0.192774  -0.205199  -0.234105
Recv:   -0.054537  -0.072290  -0.097036  -0.123261  -0.146853  -0.165101  -0.176698  -0.181737  -0.181717  -0.179537  -0.179498  -0.187304  -0.210062
Recv:   -0.042500  -0.056728  -0.081636  -0.110000  -0.136358  -0.157006  -0.170000  -0.175154  -0.174043  -0.170000  -0.168117  -0.175247  -0.200000
Recv:   -0.030988  -0.043312  -0.069229  -0.099835  -0.128425  -0.150486  -0.163704  -0.167960  -0.165332  -0.160093  -0.158712  -0.169855  -0.204383
Recv:   -0.020000  -0.034001  -0.061235  -0.092459  -0.120835  -0.141933  -0.153735  -0.156625  -0.153399  -0.149259  -0.151817  -0.171090  -0.219506
Recv:   -0.010000  -0.031759  -0.059722  -0.087500  -0.110648  -0.126667  -0.135000  -0.137037  -0.136111  -0.137500  -0.148426  -0.178056  -0.237500
Recv:   -0.001914  -0.040556  -0.067404  -0.084527  -0.094204  -0.098925  -0.101389  -0.104504  -0.111388  -0.125370  -0.149988  -0.188988  -0.246327
Recv:   0.002870  -0.065364  -0.087642  -0.083045  -0.067120  -0.051872  -0.045772  -0.053750  -0.077202  -0.113981  -0.158408  -0.201261  -0.229784
Recv:   0.002500  -0.112160  -0.124444  -0.082500  -0.024290  0.022407  0.040000  0.021080  -0.031574  -0.105000  -0.176049  -0.211389  -0.167500
Recv: 
Recv: ok
```
Settings:
[image]

[... 12 lines cut ...]
[2020-04-04 03:05:25,750] DEBUG: ok B:25.0 /0.0 T0:24.9 /0.0
[2020-04-04 03:05:30,751] DEBUG: ok B:25.0 /0.0 T0:24.7 /0.0
[2020-04-04 03:05:35,752] DEBUG: ok B:25.0 /0.0 T0:24.8 /0.0
[2020-04-04 03:05:40,753] DEBUG: ok B:25.0 /0.0 T0:24.8 /0.0
[2020-04-04 03:05:45,754] DEBUG: ok B:25.0 /0.0 T0:24.8 /0.0
[2020-04-04 03:05:50,755] DEBUG: ok B:25.0 /0.0 T0:24.8 /0.0
[2020-04-04 03:05:55,756] DEBUG: ok B:25.0 /0.0 T0:24.8 /0.0
[2020-04-04 03:06:00,757] DEBUG: ok B:25.0 /0.0 T0:24.8 /0.0
[2020-04-04 03:06:05,758] DEBUG: ok B:25.0 /0.0 T0:24.7 /0.0
[2020-04-04 03:06:10,758] DEBUG: ok B:25.0 /0.0 T0:24.8 /0.0
[2020-04-04 03:06:15,760] DEBUG: ok B:25.0 /0.0 T0:24.8 /0.0
[2020-04-04 03:06:19,591] DEBUG: Canceling mesh collection per user request
[2020-04-04 03:06:19,591] DEBUG: Mesh data collected prior to cancel:
[2020-04-04 03:06:19,592] DEBUG: []
[2020-04-04 03:06:19,592] DEBUG: Mesh data after clearing:
[2020-04-04 03:06:19,592] DEBUG: []
[2020-04-04 03:09:44,956] DEBUG: Canceling mesh collection per user request
[2020-04-04 03:09:44,957] DEBUG: Mesh data collected prior to cancel:
[2020-04-04 03:09:44,958] DEBUG: []
[2020-04-04 03:09:44,958] DEBUG: Mesh data after clearing:
[2020-04-04 03:09:44,959] DEBUG: []

```

I'm running octoprint 1.4.0 with bed visualizer 0.1.13

jneilliii, 2020-04-04:
Okay, I have an idea...set Processing Timeout to 1800, restart OctoPrint and I bet it works.

larryhan, 2020-04-04:
That works, thank you so much!

ProfEngr, 2020-04-05:
Something changed with the last update. Your suggestion before the issue was closed no longer works with a processing delay of 600. I have the following in the plugin's gcode window:

@BEDLEVELVISUALIZER
BED_MESH_PROFILE LOAD=MESH50C ; PLA mesh stored end of file
;BED_MESH_PROFILE LOAD=MESH70C ; PETG mesh stored end of file
BED_MESH_OUTPUT

It fails whether the load command is commented or not. The terminal window generates this:

```
Send: BED_MESH_OUTPUT
Recv: // Mesh Leveling Probed Z positions:
Recv: // 0.081250 0.071250 0.051250 0.055000 0.041250 0.028750 0.007500 0.001250 0.003750
Recv: // 0.033750 0.003750 -0.032500 -0.018750 -0.013750 -0.016250 -0.033750 -0.046250 -0.053750
Recv: // 0.036250 0.000000 -0.052500 -0.028750 0.010000 0.033750 0.000000 -0.033750 -0.066250
Recv: // 0.022500 -0.008750 -0.045000 -0.007500 0.032500 0.056250 0.020000 -0.016250 -0.077500
Recv: // 0.055000 0.042500 0.013750 0.018750 0.013750 0.007500 -0.040000 -0.087500 -0.123750
Recv: // 0.100000 0.073750 0.045000 0.036250 0.003750 -0.013750 -0.060000 -0.108750 -0.120000
Recv: // 0.058750 0.042500 0.011250 0.028750 0.012500 0.005000 -0.038750 -0.053750 -0.031250
Recv: // 0.066250 0.040000 0.010000 0.025000 0.007500 -0.007500 -0.050000 -0.083750 -0.076250
Recv: // 0.111250 0.083750 0.053750 0.067500 0.038750 0.016250 -0.035000 -0.067500 -0.085000
Recv: Mesh X,Y: 25,25
Recv: Search Height: 5
Recv: Mesh Average: -0.00
Recv: Mesh Range: min=-0.1254 max=0.1113
Recv: Interpolation Algorithm: bicubic
Recv: Measured points:
Recv:   0.111250  0.104157  0.092176  0.083750  0.074509  0.061157  0.053750  0.057056  0.064139  0.067500  0.060361  0.047500  0.038750  0.032491  0.023509  0.016250  0.002019  -0.020324  -0.035000  -0.045167  -0.058833  -0.067500  -0.073259  -0.080685  -0.085000
Recv:   0.099028  0.091988  0.080094  0.071722  0.062488  0.049137  0.041741  0.045085  0.052261  0.055796  0.049478  0.037968  0.030111  0.024440  0.016256  0.009556  -0.003947  -0.025217  -0.039278  -0.049641  -0.063722  -0.072398  -0.076359  -0.080780  -0.083269
Recv:   0.078806  0.071880  0.060174  0.051917  0.042692  0.029335  0.021954  0.025368  0.032707  0.036537  0.031574  0.022296  0.015917  0.011235  0.004398  -0.001361  -0.013665  -0.033177  -0.046222  -0.056863  -0.071547  -0.080185  -0.081136  -0.080540  -0.079981
Recv:   0.066250  0.059500  0.048083  0.040000  0.030778  0.017389  0.010000  0.013481  0.020963  0.025000  0.020870  0.012963  0.007500  0.003500  -0.002389  -0.007500  -0.019093  -0.037574  -0.050000  -0.060620  -0.075352  -0.083750  -0.082694  -0.078806  -0.076250
Recv:   0.062250  0.056308  0.046219  0.038926  0.029670  0.016050  0.008546  0.012307  0.020283  0.024657  0.020776  0.013230  0.008074  0.004767  -0.000039  -0.004500  -0.016116  -0.034738  -0.047046  -0.056212  -0.068491  -0.075194  -0.072506  -0.066282  -0.062343
Recv:   0.058917  0.054257  0.046277  0.040241  0.030935  0.016952  0.009259  0.013404  0.022057  0.026870  0.023158  0.015828  0.010926  0.008586  0.005329  0.001778  -0.010050  -0.029153  -0.041426  -0.048249  -0.056558  -0.060583  -0.055735  -0.046523  -0.040824
Recv:   0.058750  0.054759  0.047880  0.042500  0.033194  0.019056  0.011250  0.015361  0.023972  0.028750  0.024926  0.017435  0.012500  0.010611  0.008111  0.005000  -0.006991  -0.026426  -0.038750  -0.044491  -0.050954  -0.053750  -0.048028  -0.037639  -0.031250
Recv:   0.070500  0.065774  0.057688  0.051602  0.042504  0.028806  0.021000  0.023194  0.028558  0.031176  0.025852  0.016330  0.010102  0.007345  0.003713  -0.000083  -0.012217  -0.031802  -0.044537  -0.052562  -0.062826  -0.068250  -0.065098  -0.058391  -0.054185
Recv:   0.089917  0.083839  0.073535  0.066148  0.057400  0.044472  0.036694  0.035721  0.035715  0.034769  0.027028  0.014246  0.005926  0.001640  -0.004027  -0.009056  -0.021401  -0.041178  -0.054602  -0.066696  -0.083745  -0.093861  -0.095309  -0.095196  -0.094898
Recv:   0.100000  0.093231  0.081796  0.073750  0.065222  0.052750  0.045000  0.042231  0.039185  0.036250  0.027343  0.013046  0.003750  -0.001324  -0.008065  -0.013750  -0.026222  -0.046139  -0.060000  -0.074565  -0.095741  -0.108750  -0.113278  -0.117639  -0.120000
Recv:   0.089370  0.083627  0.073883  0.066870  0.058467  0.045986  0.038306  0.036136  0.034126  0.032065  0.024948  0.013467  0.005954  0.001720  -0.004049  -0.009204  -0.021754  -0.041957  -0.056037  -0.070734  -0.092163  -0.105611  -0.112214  -0.119943  -0.124343
Recv:   0.068907  0.064982  0.058238  0.053046  0.044770  0.032133  0.024556  0.023887  0.024464  0.024435  0.020718  0.014497  0.010324  0.007768  0.003995  -0.000046  -0.012659  -0.033259  -0.047574  -0.061891  -0.082749  -0.096250  -0.105684  -0.118137  -0.125407
Recv:   0.055000  0.052000  0.046778  0.042500  0.034176  0.021296  0.013750  0.014343  0.017102  0.018750  0.017620  0.015380  0.013750  0.012593  0.010546  0.007500  -0.005000  -0.025667  -0.040000  -0.053889  -0.074111  -0.087500  -0.098843  -0.114519  -0.123750
Recv:   0.044556  0.040306  0.033006  0.027398  0.018298  0.004526  -0.003167  -0.000214  0.006678  0.011352  0.013986  0.017427  0.019519  0.021052  0.022631  0.021824  0.010357  -0.009123  -0.022667  -0.035503  -0.054257  -0.067083  -0.080532  -0.099877  -0.111352
Recv:   0.030333  0.023773  0.012633  0.004574  -0.005862  -0.021146  -0.029139  -0.022550  -0.009272  0.000065  0.008402  0.020439  0.028176  0.033894  0.041154  0.043870  0.034113  0.016650  0.004444  -0.006695  -0.023085  -0.034944  -0.05 [... cut ...]
Recv:   0.068194  0.064335  0.057741  0.052806  0.045619  0.034813  0.028593  0.029748  0.033108  0.034935  0.032776  0.028716  0.025824  0.023069  0.018914  0.015676  0.010012  0.001247  -0.004306  -0.006996  -0.010157  -0.011954  -0.012145  -0.011973  -0.011824
Recv:   0.081250  0.078806  0.074583  0.071250  0.065417  0.056472  0.051250  0.051889  0.054083  0.055000  0.051528  0.045444  0.041250  0.037731  0.032602  0.028750  0.022648  0.013324  0.007500  0.005120  0.002574  0.001250  0.001750  0.002972  0.003750
Recv: 
Recv: ok
```

jneilliii, 2020-04-05:
@ProfEngr, did you try what @larryhan did, which fixed his?

RustyWraith, 2020-04-06:
With Processing Timeout 1800 it works only one time after saving configuration.

jneilliii, 2020-04-06:
Now I'm confused. I'll need logs please, enable debug logging in the plugin's settings, restart octoprint and upload the logs here in a comment by dragging and dropping the plugin_bedlevelvisualizer_debug.log file that you can find in the logging section of octoprint.

RustyWraith, 2020-04-06:
Works fine after a restart. "@BEDLEVELVISUALIZER" was added before the restart.
Thank you!

ProfEngr, 2020-04-11:
> 
> 
> @ProfEngr, did you try what @larryhan did, which fixed his?

Yeah, just tried it. Somehow 1800 did the trick even though it updated the mesh before I could click over to the terminal tab to view the output. 🤯 🤷‍♂️ Thanks.

AmyTheCute, 2020-04-14:
I have to restart octoprint everytime I want to update the mesh. It only works for the first time. any ideas? i set my timeout to 1800

RustyWraith, 2020-04-14:
Try:

G28
BED_MESH_CALIBRATE
@BEDLEVELVISUALIZER
BED_MESH_OUTPUT

Then system restart.

Kwadad, 2020-04-21:
> Try:
> 
> G28
> BED_MESH_CALIBRATE
> @BEDLEVELVISUALIZER
> BED_MESH_OUTPUT
> 
> Then system restart.

Tried this didn't work.  I just installed Octoprint and have been trying to get it setup.
```
G28
BED_MESH_CALIBRATE
@BEDLEVELVISUALIZER
BED_MESH_OUTPUT
```
Is my settings
Output at Terminal 
```Send: M105
Recv: ok B:20.5 /0.0 T0:21.6 /0.0
Send: G28
Recv: ok
Send: BED_MESH_CALIBRATE
Recv: // probe at 52.000,20.000 is z=2.495000
Recv: // probe at 136.000,20.000 is z=2.410000
Recv: // probe at 220.000,20.000 is z=2.335000
Recv: // probe at 220.000,120.000 is z=2.152500
Recv: // probe at 136.000,120.000 is z=2.167500
Recv: // probe at 52.000,120.000 is z=2.247500
Recv: // probe at 52.000,220.000 is z=2.122500
Recv: // probe at 136.000,220.000 is z=2.115000
Recv: // probe at 220.000,220.000 is z=1.997500
Recv: // Mesh Bed Leveling Complete
Recv: // Bed Mesh state has been saved to profile [default]
Recv: // for the current session.  The SAVE_CONFIG command will
Recv: // update the printer config file and restart the printer.
Recv: ok
Send: BED_MESH_OUTPUT
Recv: // Mesh Leveling Probed Z positions:
Recv: // 0.225000 0.140000 0.065000
Recv: // -0.022500 -0.102500 -0.117500
Recv: // -0.147500 -0.155000 -0.272500
Recv: Mesh X,Y: 7,7
Recv: Search Height: 4
Recv: Mesh Average: -0.06
Recv: Mesh Range: min=-0.2725 max=0.2250
Recv: Interpolation Algorithm: lagrange
Recv: Measured points:
Recv:   -0.147500  -0.137778  -0.140278  -0.155000  -0.181944  -0.221111  -0.272500
Recv:   -0.119444  -0.129599  -0.142654  -0.158611  -0.177469  -0.199228  -0.223889
Recv:   -0.077778  -0.102469  -0.123580  -0.141111  -0.155062  -0.165432  -0.172222
Recv:   -0.022500  -0.056389  -0.083056  -0.102500  -0.114722  -0.119722  -0.117500
Recv:   0.046389  0.008642  -0.021080  -0.042778  -0.056451  -0.062099  -0.059722
Recv:   0.128889  0.092623  0.062346  0.038056  0.019753  0.007438  0.001111
Recv:   0.225000  0.195556  0.167222  0.140000  0.113889  0.088889  0.065000
Recv: 
Recv: ok
Send: M105
Recv: ok B:20.7 /0.0 T0:21.7 /0.0
```

Kwadad, 2020-04-21:
I'm running on a R-Pi how long should mesh take?

jneilliii, 2020-04-21:
Lots of variables in that question, number of probe points, how many times it probes those points, how fast your probe moves, etc. How looking does it take for the process to complete when you type the following commands from beginning to last ok?

```
G28
BED_MESH_CALIBRATE
@BEDLEVELVISUALIZER
BED_MESH_OUTPUT
```

The default 30 minutes should be enough, but others here have reported that they had to restart octoprint after entering their gcode commands into settings before it would work.

Mide0, 2021-08-16:
I had issues with this as well. What I had to do is put the @BEDLEVELVISUALIZER first to tell the bed visualizer to look for input then the rest. Mine looks like this and works for me. I hope this helps someone.

G28
@BEDLEVELVISUALIZER
BED_MESH_CALIBRATE
BED_MESH_OUTPUT
G28