jneilliii/OctoPrint-BedLevelVisualizer #93: Configurable plot height for the Bed visualizer plot area

ikout, 2018-10-17:
Hello,

I like the plugin and it's useful, can I ask for a small enchantment if possible?
Can we add an option or where we can change that at the plugin manually to increase the height of the plot?

I have a relative large screen and I use the "Themify" plugin as well and even with or without it when I am at the BedLevelVisualizer tab the plot area is quite small especially at the vertical/height dimension if that could be adjusted it would be great!

Also the default plot area is "cropping" a bit of the graph as you can see at the screenshot below.
If this has been mentioned before and I missed it sorry.

Thanks!

[image]

[image]

jneilliii, 2018-10-20:
You could adjust your themeify to set whatever height you want. Without knowing your current config it's hard to say, but for me just swapping the tab elements width properties made the graph fill my screen. Something like this.

[image]

Produced the following.

[image]

The side effect is that all tabs get stretched and may cause other UI issues.  You could just adjust the size of the plot container and let it overflow by adding the following in Themeify (selector is #bedlevelvisualizergraph).

[image]

Produces the following.

[image]

ikout, 2018-10-21:
Thanks!
I was looking at a Themify way but I didn't had any luck finding the proper selector...

"#bedlevelvisualizergraph" and only changing the "height" value works for this change just fine.

closed 2018-10-21 by jneilliii