jneilliii/OctoPrint-BedLevelVisualizer #97: 0.1.7 offers to upgrade to 0.1.7 (!)

jollino, 2018-11-25:
Hello there,
I upgraded the plugin from 0.1.6 to 0.1.7, but I keep getting a notification that 0.1.7 is out. Re-updating just leads to the same thing. Uninstalling and reinstalling made it go away for now, but I'm sure it'll come back when OctoPrint updates its plugin list.

I *think* — I've never written plugins for OctoPrint myself so this is a wild guess — that it may be due to `setup.py` still having `plugin_version = "0.1.6"` on line 17, so the plugin system doesn't know it's actually gotten the current one. Indeed, the plugin list still claims it's version 0.1.6.

Hope this helps!

ra100, 2018-11-25:
yup, same issue, it's because version was not changed in `setup.py`

LordNikkon69, 2018-11-25:
I'm not the best in python (start to learn) but seems like that because after upgrade i got timeout function.

jneilliii, 2018-11-25:
Sorry about that guys, forgot to bump the version. Thanks @ra100 for the pull request. You'll have to reinstall 0.1.7 one more time i'm afraid to make it go away.

closed 2018-11-25 by jneilliii