jneilliii/OctoPrint-BedLevelVisualizer #99: Error while trying to install....

areonic, 2018-12-04:
Hello, I am trying to install the plugin and keep getting an error. It will seem to be normal for a few mins, then I assume stop with an error. I am running Octoprint 1.3.9 on a PI and have tried to install it from my PC and IPAD and have failed both times (doubt that matters). This is what I get at the bottom of the box, each time I try.....

Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.9/README.Bugs> for instructions.
The bug is not reproducible, so it is likely a hardware or OS problem.
arm-linux-gnueabihf-gcc: numpy/core/src/multiarray/multiarraymodule.c
error: Command "arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -DNPY_INTERNAL_BUILD=1 -DHAVE_NPY_CONFIG_H=1 -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE=1 -D_LARGEFILE64_SOURCE=1 -Ibuild/src.linux-armv7l-2.7/numpy/core/src/private -Inumpy/core/include -Ibuild/src.linux-armv7l-2.7/numpy/core/include/numpy -Inumpy/core/src/private -Inumpy/core/src -Inumpy/core -Inumpy/core/src/npymath -Inumpy/core/src/multiarray -Inumpy/core/src/umath -Inumpy/core/src/npysort -I/usr/include/python2.7 -Ibuild/src.linux-armv7l-2.7/numpy/core/src/private -Ibuild/src.linux-armv7l-2.7/numpy/core/src/npymath -Ibuild/src.linux-armv7l-2.7/numpy/core/src/private -Ibuild/src.linux-armv7l-2.7/numpy/core/src/npymath -Ibuild/src.linux-armv7l-2.7/numpy/core/src/private -Ibuild/src.linux-armv7l-2.7/numpy/core/src/npymath -c numpy/core/src/multiarray/descriptor.c -o build/temp.linux-armv7l-2.7/numpy/core/src/multiarray/descriptor.o -MMD -MF build/temp.linux-armv7l-2.7/numpy/core/src/multiarray/descriptor.o.d" failed with exit status 1
----------------------------------------
Command "/home/pi/oprint/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-H5Tq6M/numpy/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-PX3HVC-record/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/oprint/include/site/python2.7/numpy" failed with error code 1 in /tmp/pip-build-H5Tq6M/numpy/
You are using pip version 9.0.1, however version 18.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

Also, here is my log if that helps.

Thank you,
[plugin_pluginmanager_console.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/2642764/plugin_pluginmanager_console.log)

jneilliii, 2018-12-04:
Looks like your device is having issues compiling numpy (one of the dependicies of the plugin since version 0.1.3).  Did you build the pi from scratch or use an octopi distribution?  I would try to insure that all your underlying pi stuff was patched and up to date.  SSH to your pi (or use local keyboard/video) and type the following commands and see if that helps.

```
sudo apt-get update
sudo apt-get upgrade
cd ~/oprint/bin
./pip install numpy
```

Those last two lines are to try and pre-install the numpy dependency outside of the OctoPrint plugin install procedure.

The other option is to install a version of the plugin prior to the numpy dependency, assuming your firmware is supported prior to that version.  You'd have to look through the release notes of those newer versions to determine if your firmware was added before or after.

areonic, 2018-12-05:
Hello, I did exactly what you said to there, and this is what happened....

`Command "/home/pi/oprint/bin/python2 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-acuL3u/numpy/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-5HNNjd-record/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/oprint/include/site/python2.7/numpy" failed with error code 1 in /tmp/pip-build-acuL3u/numpy/
You are using pip version 9.0.1, however version 18.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
pi@octopi:~/oprint/bin $
`

Thank you

areonic, 2018-12-05:
Tried it again have updating pip and got this.....

`Command "/home/pi/oprint/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-VM3Xth/numpy/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-nfGB7R/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/oprint/include/site/python2.7/numpy" failed with error code 1 in /tmp/pip-install-VM3Xth/numpy/
pi@octopi:~/oprint/bin $
`