jneilliii/OctoPrint-BedLevelVisualizer #101: Formbot T-Rex 2+ Bed Visualization Timeout

JBest521, 2018-12-10:
Hi,

I am attempting to run this plugin with my formbot t-rex 2+, but am unable to get the "G29 T" command to output any data.

Here is my code in the plugin:
G28           ; Home XYZ.
@BEDLEVELVISUALIZER
G29 T

Here is what is showing in my terminal on octoprint:
Send: G28
Recv: echo:Active Extruder: 0
Send: G29 T
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:Active Extruder: 0
Recv: X:211.00 Y:171.00 Z:5.71 E:0.00 Count X:16880 Y:27360 Z:9136
Recv: ok
Recv: echo:busy: processing
Send: M105
Recv: echo:busy: processing
Recv: echo:busy: processing
Send: M105
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: ok
Recv: ok T:20 /0 T0:20 /0 T1:19 /0 @:0 @0:0 @1:0
Recv: ok T:20 /0 T0:20 /0 T1:19 /0 @:0 @0:0 @1:0
Send: M105
Recv: ok T:20 /0 T0:20 /0 T1:19 /0 @:0 @0:0 @1:0
Send: M105
Recv: ok T:20 /0 T0:20 /0 T1:19 /0 @:0 @0:0 @1:0
Send: M105
Recv: ok T:20 /0 T0:20 /0 T1:19 /0 @:0 @0:0 @1:0
Send: M105
Recv: ok T:20 /0 T0:20 /0 T1:19 /0 @:0 @0:0 @1:0
Send: M105
Recv: ok T:20 /0 T0:20 /0 T1:19 /0 @:0 @0:0 @1:0
Send: M105
Recv: ok T:20 /0 T0:20 /0 T1:19 /0 @:0 @0:0 @1:0
Send: M105
Recv: ok T:20 /0 T0:20 /0 T1:19 /0 @:0 @0:0 @1:0

The last two lines just continue to repeat until the plugin times out.

jneilliii, 2018-12-10:
Seems like a firmware related issue. Are you sure your firmware supports auto bed leveling and the G29 command?

JBest521, 2018-12-10:
Yes, this printer does support auto bed leveling. The factory files included UBL gcode with G28 and G29 commands, which I have successfully run.

jneilliii, 2018-12-10:
Do you know what the underlying firmware is? If you send the command `G29 T` or `G29 T1` in the terminal tab does it respond with a leveling report? The reporting mechanism in the firmware may have been left out to save space, so if we can't get a report in the terminal tab of the offsets, the plugin won't be able to visualize them.

closed 2019-01-01 by stale[bot]