jneilliii/OctoPrint-BedLevelVisualizer #104: t.glplot is undefined error; plotly.js and Firefox issues

reven, 2018-12-16:
I get an error with the plugin. It's strange because everything used to work perfectly. I think the error is related to the `plotly.js`library on Firefox, because when I go to the [example page](https://plot.ly/javascript/3d-surface-plots/) I can't see any of the plots; they all say "WebGL is not supported by your browser...". But it is, a quick visit to https://get.webgl.org/ makes that clear.

Safari works perfectly on the same machine, same Octopi instance, same config...

Below is the error. I'm running Firefox 64.0 on a macbook Pro.
    Bed Visualizer Error
    Looks like your settings are not correct or there was an error. Please see the Readme for configuration hints.

    TypeError: t.glplot is undefined

If this would be useful to troubleshoot, I'd be glad to give you more debug info; just ask.

jneilliii, 2018-12-16:
Maybe webgl was diagnosed in an update to firefox? You could try forcing it on...

https://www.sitepoint.com/firefox-enable-webgl-blacklisted-graphics-card/

closed 2018-12-16 by jneilliii

Vertabreak, 2019-12-13:
enabled webgl as suggested problem still exists this started after a update that i got today.