jneilliii/OctoPrint-BedLevelVisualizer #108: failed with error code 1 in /tmp/pip-build-NJvjOD

shacker2, 2019-02-11:
i getting this error

failed with error code 1 in /tmp/pip-build-NJvjOD/

and cant install teh plugin. i try to upgrade pyton, etc. but same result

jneilliii, 2019-02-11:
That's a numpy install/build error. I'd recommend to try installing numpy ahead of time if possible. Ssh to your pi, and run these commands. 

```
cd oprint/bin
pip install numpy
```

Once that runs through its install/build process the plugin should install without any issues. The other option is to install an older version that doesn't have that dependency on numpy.

shacker2, 2019-02-11:
thanks for the response

but i get this error when install numpy
Failed building wheel for numpy

Failed cleaning build dir for numpy

i check that error

Skybeat, 2019-02-23:
I have exactly the same issue like shacker2.
I have try differend thinks with a clean install of octopi as base.
No chance to get BedLevelVisualizer installed :-(

shacker2, 2019-02-24:
i updated everything, added  python3, etc nothing help

jneilliii, 2019-02-24:
Python3 wouldn't help until a future release of octoprint that supports it. My only recommendation is to search for those errors related to installing numpy on a pi, or install an older version of the plugin that doesn't have the numpy requirement. My best guess is that your pi doesn't have the tools necessary for compiling, sorry I'm not much help.

Skybeat, 2019-02-24:
After 2 days trying to install BedLevelVisualizer plugin in any way, I must say that BedLevelVisualizer plugin is currently not working / usable with OctoPi, because of the issues with the installation and depency of numpy.
That's too bad because I liked BedLevelVisualizer.

jneilliii, 2019-02-24:
As developer of ther plugin I'm sorry to hear you have so many problems. I haven't ever had an issue installing and have done so even on the pi zero w running octopi 0.15.1, it just took an extremely long time. As mentioned before you can install a non numpy version if you want by grabbing the url for the version prior to that dependency.

Skybeat, 2019-02-24:
@jneilliii 
How do you install the plugin after a fresh installation of octopi (latest version 0.16.0)?
I have try it with a fresh installation of OctoPi 0.15.1 as well as 0.16.0.
I have try it via OctoPi UI (PacketManager) and direct via SSH and pip. 
The plugin try to load and install numpy 1.1.6.1, have you the same version installed on your pi?

Have you try to install your plugin on a Pi 3B ?

jneilliii, 2019-02-24:
Maybe that's the issue. The numpy version you list is different than the one that installed on my pi when I had done this before. Let me test with the latest octopi from scratch.

Skybeat, 2019-02-24:
thank you for the fast replay.
what kind of numpy version do you have installed?
And how can i change the version of numpy what will be installed within the installation of your plugin?

And a huge thanks to your fast support!

jneilliii, 2019-02-24:
I honestly don't know what the pip command would be to install a specific version, but the version I have installed is 1.15.4 within my octoprint instance.  When you try to install you have to make sure to do it from the correct folder within the pi.  So when you SSH make sure you `cd ~/oprint/bin` and then run the command `./pip install numpy`. That ensures your installing in the python 2.7 environment that the plugin installs from.

Skybeat, 2019-02-24:
@jneilliii 
the command is `pip install numpy==1.15.4`. I will try this now and let you know the outcome.

Skybeat, 2019-02-24:
I have some new.
The first try to install numpy 1.15.4 failed again. (via `pip install numpy==1.15.4`)
After that, i have upgrade setuptools of pip via `./pip install --upgrade setuptools` and installed numpy 1.15.4 again. 
This time the installation was fine.

In the last step i have installed your plugin via `pip install "https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip"` and i worked also very well now.

U hope this will help all other useres with the issue of numpy installation.

jneilliii, 2019-02-24:
That's awesome news.  I just started the fresh install of the pugin on a brand new octopi 0.16.0 instance.  I'll report back here as well.

Skybeat, 2019-02-24:
the installation of numpy 1.15.4 takes between 8 to 10 minutes for me.

jneilliii, 2019-02-24:
Yeah, so the install worked for me from the plugin manager without any issues.  Maybe it's because of my process.  The very first thing I do after installing octopi on a new sd card is a `sudo apt-get update` `sudo apt-get upgrade` and maybe that was updating setuptools somehow. Hopefully the upgrade of setuptools will help with @shacker2 issue.

closed 2019-03-17 by stale[bot]

Kiron21, 2019-05-15:
If you have only 512MB of RAM you can try my solution:

1) stop Octoprint: /etc/init.d/octoprint stop
2) stop other services like mjpeg-streamer the same way.
3) instal numpy (example): /home/octoprint/OctoPrint/venv/bin/python -m pip install numpy -v

adrianmihalko, 2019-06-23:
I am on Pi 3 and I have same problem. Is there any way to install this plugin?

Log:
https://pastebin.com/mHQv56Tg

bonkas, 2020-06-06:
I just wanted to share my results with this plugin - I was having issues getting it installed, pages and pages of errors during compile of numpty. This was on a fresh install of OctoPrint 1.4.0 running on OctoPi 0.17.0.

As per previous post by @jneilliii `./pip install --upgrade setuptools` solved the issues and I could install numpy manually and proceed with the plugin install.

jneilliii, 2020-06-06:
@bonkas thanks for reporting. What device were you installing on?

bonkas, 2020-06-08:
> 
> 
> @bonkas thanks for reporting. What device were you installing on?

I am using a RPi Model 3B

jneilliii, 2020-06-09:
Thanks, it has been brought to my attention recently that RPi Model A+ devices can't compile numpy because they run out of memory, which is why I was asking.

Kiron21, 2020-06-10:
The solution is to use the swap to increase your RAM memory.

I´ve got only 512MB ;)

https://github.com/rezaxdi/tensorflow-on-orangepi-zero/blob/master/GUIDE.md

Trevor-Miller, 2020-09-28:
Confirmed that on a fresh install of Octopi, I was getting this issue when installing through the plugin repo, but running `./pip install --upgrade setuptools` allowed me to install numpy and everything moved forward smoothly

eddiesomesan, 2021-01-24:
I had the same problem. I was unable to install bed visualizer on a RPi 3 A+ on a clean image.  And I also think it was because of RAM run out. Numpy got stucked after a while. I was able to reinstall numpy only after stopping the octoprint service: /etc/init.d/octoprint stop
After that, I have installed Bed Visualizer: ./pip install https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip
If it still doesn`t work, before this, try upgradinng setuptools: ./pip install --upgrade setuptools
All the commands were executed from /home/pi/oprint/bin
Thank you for your help.

> Thanks, it has been brought to my attention recently that RPi Model A+ devices can't compile numpy because they run out of memory, which is why I was asking.

> If you have only 512MB of RAM you can try my solution:
> 
> 1. stop Octoprint: /etc/init.d/octoprint stop
> 2. stop other services like mjpeg-streamer the same way.
> 3. instal numpy (example): /home/octoprint/OctoPrint/venv/bin/python -m pip install numpy -v