jneilliii/OctoPrint-BedLevelVisualizer #111: Graph not plotting correctly

Webstas, 2019-03-04:
For  some reason i keep getting these weird results
[image]

I am using Marlin 1.1.8 and the 3 point ABL setting

In the firmware i have it set to probe
```
  // 3 arbitrary points to probe.
  // A simple cross-product is used to estimate the plane of the bed.
  #define ABL_PROBE_PT_1_X 15
  #define ABL_PROBE_PT_1_Y 150
  #define ABL_PROBE_PT_2_X 15
  #define ABL_PROBE_PT_2_Y 20
  #define ABL_PROBE_PT_3_X 160
  #define ABL_PROBE_PT_3_Y 100
```
This is the terminal output
```
Send: M155 S30
Recv: ok
Send: G29 T
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: X:187.00 Y:147.01 Z:10.36 E:0.00 Count X:14960 Y:11760 Z:4000
Recv: ok
Send: G81
Recv: ok
Send: M420 V
Recv: Bed Level Correction Matrix:+1.000000 +0.000000 -0.000152
Recv: +0.000000 +1.000000 +0.001019
Recv: +0.000152 -0.001019 +1.000000
Recv: echo:Bed Leveling On
Recv: ok
Send: M155 S3
Recv: ok
```

jneilliii, 2019-03-05:
I see why it's doing it, not sure right now how to adjust for it though. If you send a command G29 T1 in terminal what is reported?

Webstas, 2019-03-05:
Thank you for the help, this is what is returned

Send: M155 S30
Recv: ok
Send: G29 T1
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: X:186.99 Y:147.01 Z:10.48 E:0.00 Count X:14960 Y:11760 Z:4000
Recv: ok
Send: G81
Recv: ok
Send: M420 V
Recv: Bed Level Correction Matrix:+1.000000 +0.000000 -0.000837
Recv: +0.000001 +0.999999 +0.001254
Recv: +0.000837 -0.001254 +0.999999
Recv: echo:Bed Leveling On
Recv: ok
Send: M155 S3
Recv: ok

jneilliii, 2019-03-18:
Yeah, so looking at the graph it is rendering the points that you are getting returned from the firmware.
```
Recv: Bed Level Correction Matrix:+1.000000 +0.000000 -0.000152
Recv: +0.000000 +1.000000 +0.001019
Recv: +0.000152 -0.001019 +1.000000
```

You'll see from that report that on the first line the first point is 1mm above, on the second line the middle point is 1mm above, and on the third line the third point is 1mm above. That indicates to me that you have a bow diagonally down the center of your bed where possibly the other corners are being tightened too tight, or the middle of your bed is being pushed up.  The plugin is rendering correctly what is there in the Bed Leveling Correction Matrix report.

closed 2019-03-18 by jneilliii