jneilliii/OctoPrint-BedLevelVisualizer #112: Unknown command: @BEDLEVELVISUALIZER

Caliban2017, 2019-03-05:
Hello,

i installed your Plugin to my Octoprint and it works great so far. I use an Ender 3 mit Marlin 2.X Firmware, when i use the Update Button in the Plugin, the Commands G28 and G29 are working to get the Mesh Data from the BLTouch. But i want to show the Data before every Print and for that i put in my GCODE Startscript the Command @BEDLEVELVISUALIZER and after that the G29T. It does the Auto-Home before but after that  i get that Error in the Terminal: echo:Unknown command: "@BEDLEVELVISUALIZER". After that it do the Auto Leveleling and begin the Print. But the Mesh Data is not reported to the Plugin and it still shows the last Data from an manual Auto Leveling. 

Am i'm doing anything wrong? My prints are started from the Printers Display as SD-Card Print. Also a start from Octoprint does not work.

Thanks for your Help!

jneilliii, 2019-03-05:
In theory that should work fine for non-sd prints. I'm not certain but I believe for it to work it has to go through a local octoprint stored file for the gcode commands that trigger the plugin to go through the queue being monitored by octoprint. From my understanding that doesn't happen with sd prints.

Caliban2017, 2019-03-06:
Thank you for your Help. So there is no way to transmit the Data to the Plugin, when i print from SD Card? Unfortunately there is no way to simply paste the Mesh Data into the Plugin. I can see the Mesh Data Report in the Terminal before the Print starts, but the Plugin does not see it. 

Make the Plugin unfortunately useless for me, because i print always from SD Card. But thank you for your Help.

closed 2019-03-06 by jneilliii