jneilliii/OctoPrint-BedLevelVisualizer #113: Can't install it on OctoPrint 1.3.10 / OctoPi 0.16.0

Brainc0re, 2019-03-05:
This is may first octopi and all installed plugin work. But im not able to install Bed Level Visualizer.

I tried it so many times (download via plugin manager and direct zip) today without any success. I always get a unknown error...

Installing plugin "Bed Level Visualizer" from https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip...
/home/pi/oprint/bin/python2 -m pip install https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip --no-cache-dir
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip
Downloading https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from Bed-Visualizer==0.1.7) (1.3.10)
Collecting numpy (from Bed-Visualizer==0.1.7)
Downloading https://files.pythonhosted.org/packages/cf/8d/6345b4f32b37945fedc1e027e83970005fc9c699068d2f566b82826515f2/numpy-1.16.2.zip (5.1MB)
Requirement already satisfied: watchdog<0.9,>=0.8.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.7) (0.8.3)
Requirement already satisfied: python-dateutil<2.7,>=2.6.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.7) (2.6.1)
Requirement already satisfied: feedparser<5.3,>=5.2.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.7) (5.2.1)
Requirement already satisfied: requests<3,>=2.18.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.7) (2.21.0)
Requirement already satisfied: chainmap<1.1,>=1.0.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.7) (1.0.2)
Requirement already satisfied: flask<0.11,>=0.10.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.7) (0.10.1)
Requirement already satisfied: scandir<1.4,>=1.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.7) (1.3)
Requirement already satisfied: monotonic<1.4,>=1.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.7) (1.3)
Requirement already satisfied: frozendict<1.3,>=1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.7) (1.2)
Requirement already satisfied: Jinja2<2.9,>=2.8.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.7) (2.8.1)
Requirement already satisfied: emoji<0.5,>=0.4.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.7) (0.4.5)
Requirement already satisfied: netifaces<0.11,>=0.10.6 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.7) (0.10.9)
Requirement already satisfied: Flask-Babel<0.10,>=0.9 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.7) (0.9)
Requirement already satisfied: pkginfo<1.5,>=1.4.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.7) (1.4.2)
Requirement already satisfied: markdown<2.7,>=2.6.9 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.7) (2.6.11)
Requirement already satisfied: pyserial<3.5,>=3.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.7) (3.4)
Requirement already satisfied: semantic-version<2.5,>=2.4.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.7) (2.4.2)
Requirement already satisfied: werkzeug<0.9,>=0.8.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.7) (0.8.3)
Requirement already satisfied: websocket-client<0.41,>=0.40.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.7) (0.40.0)
Requirement already satisfied: psutil<5.5,>=5.4.6 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.7) (5.4.8)
Requirement already satisfied: awesome-slugify<1.7,>=1.6.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.7) (1.6.5)
Requirement already satisfied: Flask-Login<0.3,>=0.2.11 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.7) (0.2.11)
Requirement already satisfied: Flask-Principal<0.4,>=0.3.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.7) (0.3.5)
Requirement already satisfied: sarge==0.1.5post0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.7) (0.1.5.post0)
Requirement already satisfied: pylru<1.1,>=1.0.9 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.7) (1.0.9)
Requirement already satisfied: Flask-Assets<0.11,>=0.10 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.7) (0.10)
Requirement already satisfied: netaddr<0.8,>=0.7.19 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.7) (0.7.19)
Requirement already satisfied: future<0.16,>=0.15.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.7) (0.15.2)
Requirement already satisfied: Click<6.3,>=6.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.7) (6.2)
Requirement already satisfied: regex!=2018.11.6 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.7) (2019.1.24)
Requirement already satisfied: wrapt<1.11,>=1.10.11 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.7) (1.10.11)
Requirement already satisfied: futures<3.2,>=3.1.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.7) (3.1.1)
Requirement already satisfied: PyYAML<3.11,>=3.10 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.7) (3.10)
Requirement already satisfied: tornado==4.5.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.7) (4.5.3)
Requirement already satisfied: rsa<3.3,>=3.2.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.7) (3.2.3)
Requirement already satisfied: argh>=0.24.1 in /home/pi/oprint/lib/python2.7/site-packages (from watchdog<0.9,>=0.8.3->OctoPrint->Bed-Visualizer==0.1.7) (0.26.2)
Requirement already satisfied: pathtools>=0.1.1 in /home/pi/oprint/lib/python2.7/site-packages (from watchdog<0.9,>=0.8.3->OctoPrint->Bed-Visualizer==0.1.7) (0.1.2)
Requirement already satisfied: six>=1.5 in /home/pi/oprint/lib/python2.7/site-packages (from python-dateutil<2.7,>=2.6.1->OctoPrint->Bed-Visualizer==0.1.7) (1.12.0)
Requirement already satisfied: urllib3<1.25,>=1.21.1 in /home/pi/oprint/lib/python2.7/site-packages (from requests<3,>=2.18.4->OctoPrint->Bed-Visualizer==0.1.7) (1.24.1)
Requirement already satisfied: idna<2.9,>=2.5 in /home/pi/oprint/lib/python2.7/site-packages (from requests<3,>=2.18.4->OctoPrint->Bed-Visualizer==0.1.7) (2.8)
Requirement already satisfied: chardet<3.1.0,>=3.0.2 in /home/pi/oprint/lib/python2.7/site-packages (from requests<3,>=2.18.4->OctoPrint->Bed-Visualizer==0.1.7) (3.0.4)
Requirement already satisfied: certifi>=2017.4.17 in /home/pi/oprint/lib/python2.7/site-packages (from requests<3,>=2.18.4->OctoPrint->Bed-Visualizer==0.1.7) (2018.11.29)
Requirement already satisfied: itsdangerous>=0.21 in /home/pi/oprint/lib/python2.7/site-packages (from flask<0.11,>=0.10.1->OctoPrint->Bed-Visualizer==0.1.7) (1.1.0)
Requirement already satisfied: MarkupSafe in /home/pi/oprint/lib/python2.7/site-packages (from Jinja2<2.9,>=2.8.1->OctoPrint->Bed-Visualizer==0.1.7) (1.1.0)
Requirement already satisfied: Babel>=1.0 in /home/pi/oprint/lib/python2.7/site-packages (from Flask-Babel<0.10,>=0.9->OctoPrint->Bed-Visualizer==0.1.7) (2.6.0)
Requirement already satisfied: speaklater>=1.2 in /home/pi/oprint/lib/python2.7/site-packages (from Flask-Babel<0.10,>=0.9->OctoPrint->Bed-Visualizer==0.1.7) (1.3)
Requirement already satisfied: Unidecode<0.05,>=0.04.14 in /home/pi/oprint/lib/python2.7/site-packages (from awesome-slugify<1.7,>=1.6.5->OctoPrint->Bed-Visualizer==0.1.7) (0.4.21)
[... cut ...]
Running setup.py install for numpy: still running...
Running setup.py install for numpy: still running...
Running setup.py install for numpy: still running...
Running setup.py install for numpy: still running...
Error!
Could not install plugin, unknown error, please consult octoprint.log for details

After that, my Pi freeze and i have to un- and replug it. When i take a look in the log, i (as a noob) can see anything about it.

log: https://drive.google.com/open?id=1XS3EKfBnyYAqhbc3-llYxYSteek4-McY

Any ideas why i cant install it?

jneilliii, 2019-03-05:
The issue appears to be because the dependency, numpy,  is failing to complete download/install. You may want to try installing that ahead of the plugin by ssh to your pi and running the following commands. 

```
cd ~/oprint/bin
pip install numpy
```

Wait a long time for numpy to download and build. If that still fails you may want to look into #108 for tips on updating setuptools.

Brainc0re, 2019-03-05:
I will try it. But i use my 3 B+ for it when i read "Building numpy on a Raspberry Pi 1 takes about 2 hours 30 minutes" in the www and ~20min with a Pi 3

Brainc0re, 2019-03-05:
Can't edit my last post.... But

Installing collected packages: Bed-Visualize
Running setup.py install for Bed-Visualizer: started
Running setup.py install for Bed-Visualizer: finished with status 'done'
Successfully installed Bed-Visualizer-0.1.7
Done!

It worked xD Thank you

jneilliii, 2019-03-05:
Yeah, I've done it on a single core pi zero w but it did take forever. Once octoprint goes to python 3 compatibility I think precompiled install of numpy is possible.

closed 2019-03-18 by jneilliii