jneilliii/OctoPrint-BedLevelVisualizer #114: Compatibility with Monoprice Mini Delta

Jozer99, 2019-03-12:
This plugin doesn't seem to work with the Monoprice Mini Delta.  I am including bed level output below, in case you can add support.  The format looks pretty straightforward.  

Send: N43 G29 P5 V4*34
--
Recv: G29 Auto Bed Leveling
Recv: Bed X: -25.000 Y: -50.000 Z: -0.126
Recv: Bed X: -25.000 Y: -50.000 Z: -0.121
Recv: Bed X: 0.000 Y: -50.000 Z: -0.783
Recv: Bed X: 0.000 Y: -50.000 Z: -0.778
Recv: Bed X: 25.000 Y: -50.000 Z: -0.520
Recv: Bed X: 25.000 Y: -50.000 Z: -0.524
Recv: Bed X: 50.000 Y: -25.000 Z: -1.090
Recv: Bed X: 50.000 Y: -25.000 Z: -1.085
Recv: Bed X: 25.000 Y: -25.000 Z: -1.177
Recv: Bed X: 25.000 Y: -25.000 Z: -1.173
Recv: Bed X: 0.000 Y: -25.000 Z: -0.880
Recv: Bed X: 0.000 Y: -25.000 Z: -0.884
Recv: Bed X: -25.000 Y: -25.000 Z: -0.259
Recv: Bed X: -25.000 Y: -25.000 Z: -0.254
Recv: Bed X: -50.000 Y: -25.000 Z: 0.458
Recv: Bed X: -50.000 Y: -25.000 Z: 0.463
Recv: Bed X: -50.000 Y: 0.000 Z: 0.046
Recv: Bed X: -50.000 Y: 0.000 Z: 0.051
Recv: Bed X: -25.000 Y: 0.000 Z: -0.532
Recv: Bed X: -25.000 Y: 0.000 Z: -0.536
Recv: Bed X: 0.000 Y: 0.000 Z: -1.118
Recv: Bed X: 0.000 Y: 0.000 Z: -1.113
Recv: Bed X: 25.000 Y: 0.000 Z: -1.329
Recv: Bed X: 25.000 Y: 0.000 Z: -1.324
Recv: Bed X: 50.000 Y: 0.000 Z: -1.478
Recv: Bed X: 50.000 Y: 0.000 Z: -1.464
Recv: Bed X: 50.000 Y: 25.000 Z: -2.161
Recv: Bed X: 50.000 Y: 25.000 Z: -2.130
Recv: Bed X: 25.000 Y: 25.000 Z: -1.583
Recv: Bed X: 25.000 Y: 25.000 Z: -1.579
Recv: Bed X: 0.000 Y: 25.000 Z: -1.138
Recv: Bed X: 0.000 Y: 25.000 Z: -1.124
Recv: Bed X: -25.000 Y: 25.000 Z: -0.648
Recv: Bed X: -25.000 Y: 25.000 Z: -0.652
Recv: Bed X: -50.000 Y: 25.000 Z: -0.622
Recv: Bed X: -50.000 Y: 25.000 Z: -0.626
Recv: Bed X: -25.000 Y: 50.000 Z: -0.745
Recv: Bed X: -25.000 Y: 50.000 Z: -0.740
Recv: Bed X: 0.000 Y: 50.000 Z: -0.728
Recv: Bed X: 0.000 Y: 50.000 Z: -0.732
Recv: Bed X: 25.000 Y: 50.000 Z: -1.577
Recv: Bed X: 25.000 Y: 50.000 Z: -1.572
Recv: 0.600 -0.126 -0.783 -0.524 -0.701
Recv: 0.458 -0.259 -0.884 -1.177 -1.090
Recv: 0.046 -0.536 -1.118 -1.329 -1.478
Recv: -0.626 -0.652 -1.138 -1.583 -2.161
Recv: -0.758 -0.745 -0.732 -1.577 -2.422

jneilliii, 2019-03-12:
Does it not send an ok at the end? That format should work I think. How are your settings configured?

Jozer99, 2019-03-15:
Yes, I didn't paste the "ok" line, but it is there.  I am attaching what the plot generated the plugin looks like after running a level command.  The command I run is 
G28
G29 P5 V4
G28

Plotting the same XYZ data in a different program, I get something much more like the expected result.  It seems like somehow your plugin isn't passing the data to the plotting mechanism properly.  

[image]

jneilliii, 2019-03-18:
I think what it might be doing is a combination of reading the individual lines and the final output, which is causing it to mess up.  What do you get in terminal if you just send the command G29 P5?

Jozer99, 2019-03-25:
Here is the output from just a G29 P5
Send: N480298 G29 P5*104
Recv: ok N480298 P15 B14
Send: N480299 M105*25
Recv: G29 Auto Bed Leveling
Communication timeout while idle, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N480300 M105*24
Recv: -1.328 -1.504 -1.679 -1.063 -0.872
Recv: -2.057 -1.488 -1.584 -1.396 -1.260
Recv: -1.363 -1.481 -1.708 -1.455 -1.648
Recv: -3.597 -2.936 -2.043 -1.312 -1.426
Recv: -4.163 -1.912 -1.388 -1.734 -1.203
Recv: ok N480300 P15 B13

jneilliii, 2019-03-29:
Ok looks like you would benefit from adding the G29 command to the long running commands in OctoPrint settings.  

[image]

Then make sure the plugins settings have a long timeout value to prevent it from processing. Other than that I think you could use the following in the GCODE settings for the plugin and it should work unless the part `ok N480300 P15 B13` is not detected correctly.  Not sure what those last bits of text are after the work `ok`.
```
@BEDLEVELVISUALIZER
G29 P5
```

Jozer99, 2019-03-30:
G29 is already set as one of the long running commands by default in Octoprint (at least it was already there, and I didn't add it).  The G29 P5 command you recommend does not include the V4 thing, does this mean your plugin doesn't need the extra information provided with V4?

jneilliii, 2019-03-30:
Correct the verbosity level of 4 is not necessary for the plugin.

closed 2019-04-20 by stale[bot]