jneilliii/OctoPrint-BedLevelVisualizer #115: TypeError: t.glplot is undefined

rebeltaz, 2019-03-13:
When I went to the BedAnalyzer tab the first time, I clicked on "Update Mesh". It went through the leveling procedure, but then reported back with this error when it was through:

`TypeError: t.glplot is undefined`

This is what I am using for gcode:

`G28
M155 S30
@BEDLEVELVISUALIZER
G29 T
M155 S3`

jneilliii, 2019-03-13:
What browser are you using?

rebeltaz, 2019-03-13:
Firefox 65.0.1 (64-bit) under Ubuntu 18.04 LTS

rebeltaz, 2019-03-17:
Not to be a bug, but any ideas yet?

jneilliii, 2019-03-17:
Sorry, didn't see your reply before. The only thing I can think of is that your machines video card is not open gl compatible. If you go to the plotly js examples for surface charts does it load there? 

https://plot.ly/javascript/3d-surface-plots/

rebeltaz, 2019-03-17:
gl... the bane of my existence! OK.. yeah. That would be the issue. I need a new computer. Ill try that on a different system. Thank you for your help.

closed 2019-03-17 by rebeltaz