jneilliii/OctoPrint-BedLevelVisualizer #116: Version 1.8 no longer working.

DotNetRob, 2019-03-19:
Updated to version 1.8 on my Ender 3, bed visualizer keeps saying it timed out, I increased timeout and it still says it timed out.  Probing finishes and bed visualizer just keeps spinning and then times out.  Can't get new mesh data.

barko33, 2019-03-19:
I'm having the same issue.  Was working like a champ before with my Ender 3.  After upgrade to 1.8 I get a timeout and error.  Changing the timeout value has not effect.

DotNetRob, 2019-03-19:
I can see in the terminal the probe is returning the data, I am guessing the visualizer is not detecting it anymore.  Can manually get the values and put into plot.ly for now.

jneilliii, 2019-03-19:
I will revert the version for now.  Could you guys please post the results of running the commands manually in the terminal tab? And check your octoprint.log for errors. You can install the previous version using the following url in the plugin manager.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.7.zip

DotNetRob, 2019-03-19:
The only error I see in the octoprint.log is attached.  I see if a few times in the log today, but i don't think they are related to bed visualizer, but I could be wrong.

Also attached my terminal output from running a probing.  I am running 5x5 on an Ender 3 with a EZABL with the lastest TH3D firmware.  

[Error.txt](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/2981347/Error.txt)
[Terminal.txt](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/2981348/Terminal.txt)

jneilliii, 2019-03-19:
Thanks, the error is unrelated. If you haven't already reinstalled the old version do you mind checking in the plugin's settings and ser if checking the box for bilinear fixes the issue.

DotNetRob, 2019-03-19:
Still running 1.8, will test as soon as current print job is done.

DotNetRob, 2019-03-19:
Yep, turning on Bilinear Mesh Leveling works on 1.8.

barko33, 2019-03-19:
I'm running and Ender3 with a BLTouch and the latest Marlin firmware.  I haven't reverted to 1.7 yet so I'll test the Bilinear Mesh tonight.

barko33, 2019-03-20:
So 1.8 with a BLtouch and the latest Marlin firmware works with the Bilinear Mesh.

barko33, 2019-03-20:
Oops..... Works!

closed 2019-04-10 by stale[bot]