jneilliii/OctoPrint-BedLevelVisualizer #117: smoothieware issue

hclande, 2019-03-19:
i get this error when trying to update  the visualizer:
TypeError: Plotly.react is not a function

I am using smoothieboard, the data in stored data tab is showing valid data points.

output from terminal after probing:

Recv: Rectangular Grid Probe...
Send: M105
Recv: Probe start ht is 5.000000 mm, rectangular bed width 290.000000mm, height 185.000000mm, grid size is 7x5
Recv: probe at 0,0 is 0.002810 mm
Recv: DEBUG: X0.0000, Y0.0000, Z-0.0066
Send: M105
Recv: DEBUG: X48.3333, Y0.0000, Z0.1095
Recv: DEBUG: X96.6667, Y0.0000, Z0.1200
Send: M105
Recv: DEBUG: X145.0000, Y0.0000, Z0.1136
Recv: DEBUG: X193.3333, Y0.0000, Z0.0571
Send: M105
Recv: DEBUG: X241.6667, Y0.0000, Z-0.0679
Recv: DEBUG: X290.0000, Y0.0000, Z-0.2516
Recv: DEBUG: X290.0000, Y46.2500, Z-0.3497
Send: M105
Recv: DEBUG: X241.6667, Y46.2500, Z-0.1370
Recv: DEBUG: X193.3333, Y46.2500, Z0.0098
Send: M105
Recv: DEBUG: X145.0000, Y46.2500, Z0.0502
Recv: DEBUG: X96.6667, Y46.2500, Z-0.0142
Recv: DEBUG: X48.3333, Y46.2500, Z-0.0742
Recv: DEBUG: X0.0000, Y46.2500, Z-0.3526
Recv: DEBUG: X0.0000, Y92.5000, Z-0.6550
Recv: DEBUG: X48.3333, Y92.5000, Z-0.2797
Recv: DEBUG: X96.6667, Y92.5000, Z-0.0376
Recv: DEBUG: X145.0000, Y92.5000, Z0.0448
Recv: DEBUG: X193.3333, Y92.5000, Z-0.0208
Recv: DEBUG: X241.6667, Y92.5000, Z-0.1247
Recv: DEBUG: X290.0000, Y92.5000, Z-0.4366
Recv: DEBUG: X290.0000, Y138.7500, Z-0.4018
Recv: DEBUG: X241.6667, Y138.7500, Z-0.1402
Recv: DEBUG: X193.3333, Y138.7500, Z0.0013
Recv: DEBUG: X145.0000, Y138.7500, Z0.1155
Recv: DEBUG: X96.6667, Y138.7500, Z-0.0177
Recv: DEBUG: X48.3333, Y138.7500, Z-0.3359
Recv: DEBUG: X0.0000, Y138.7500, Z-0.7478
Recv: DEBUG: X0.0000, Y185.0000, Z-1.7402
Recv: DEBUG: X48.3333, Y185.0000, Z-0.4189
Recv: DEBUG: X96.6667, Y185.0000, Z-0.1698
Recv: DEBUG: X145.0000, Y185.0000, Z-0.0653
Recv: DEBUG: X193.3333, Y185.0000, Z-0.0227
Recv: DEBUG: X241.6667, Y185.0000, Z-0.1487
Recv: DEBUG: X290.0000, Y185.0000, Z-0.4419
Recv: -0.0066 0.1095 0.1200 0.1136 0.0571 -0.0679 -0.2516
Recv: -0.3526 -0.0742 -0.0142 0.0502 0.0098 -0.1370 -0.3497
Recv: -0.6550 -0.2797 -0.0376 0.0448 -0.0208 -0.1247 -0.4366
Recv: -0.7478 -0.3359 -0.0177 0.1155 0.0013 -0.1402 -0.4018
Recv: -1.7402 -0.4189 -0.1698 -0.0653 -0.0227 -0.1487 -0.4419
Recv: Probe completed

jneilliii, 2019-03-19:
That error is typically an issue with a conflicting plugin that also uses plotly but is an older version. What other plugins do you have installed?

hclande, 2019-03-20:
thanks! I disabled a few plugins and now it is working fine. 
plugins that i disabled:
OctoPrint-Tempsgraph
Printer Statistics
Emergency STOP! button

closed 2019-03-20 by hclande

jneilliii, 2019-03-20:
Tempsgraph iis the one I think. I think he released a version update that had the newer plotly.

jneilliii, 2019-04-09:
Tempsgraph I think uses plotly, but the author released an update that I thought put the plotly versions at the same level.