jneilliii/OctoPrint-BedLevelVisualizer #119: Visualizer does not update the values

Brainc0re, 2019-03-20:
I dont know how long this problem exist. But the today in installed a mirror on my bed. So i pressed the "update" button, it measures the bed, but the graphic does not update. Under the graphic is to read "Using saved mesh data from 6.3.2019, 00:21:27."

I also did a G28 -> G29 -> M500

The visualizer does not update and i dont know if the printer uses these old values or the new ones when i do a G29....

How can i delete these values from 6.3.2019 to force an update?

jneilliii, 2019-03-20:
The printer itself will always use the real information that was calculated by the last leveling process.  The plugin only displays either the stored data or the live data based on the settings you pick.  What do your settings look like?

Brainc0re, 2019-03-20:
G29 T
G81
M420 V

Are the command in the visualizer settings

But when i do G28-G29-M500 via the terminal. The values in the visualizer do not update.

jneilliii, 2019-03-20:
Change your settings to this and let me know what happens when you press the update button. 

```
@BEDLEVELVISUALIZER
G29 T
```

Brainc0re, 2019-03-20:
It sill show "Using saved mesh data from 6.3.2019, 00:21:27." and the same graphic :/

jneilliii, 2019-03-20:
What firmware are you using?

Brainc0re, 2019-03-20:
Marlin 1.1.9 Creality CR10-Pro

I only changed a few settings to use a 12864 Display instead of that useless touchscreen

jneilliii, 2019-03-20:
Please screenshot the plugin's settings for me.

Brainc0re, 2019-03-20:
https://imgur.com/a/8z27Wlc

Brainc0re, 2019-03-21:
Strange. I uninstalled the plug -> reboot (power off) -> reboot -> reinstalled -> reboot (regular) and the plugin showed the same picture. 

But i added the following from your github site 

`G28	
M155 S30	
@BEDLEVELVISUALIZER	
G29 T	
M155 S3`

And its working now... why? I don't know

jneilliii, 2019-03-24:
I'm not sure.  Glad it's working for you now.

closed 2019-03-24 by jneilliii