jneilliii/OctoPrint-BedLevelVisualizer #120: Tooltips on graph show negative numbers as positive (release 0.1.7)

mtwomey, 2019-04-13:
z: 0.479 shown in the image is actually -0.479 (in the data)

[image]

jneilliii, 2019-04-13:
That's strange.  Can you share your plugin settings, firmware and firmware version, and the output of running the leveling report in the terminal tab? You can see from the screenshot below it's showing negatives for me.

[image]

mtwomey, 2019-04-15:
Yep Yep. I'm running Marlin 1.1 bugfix branch. Octiprint 1.3.10, Bed Visualizer 0.1.7 (with it's default settings).

Here is my bed data as displayed in the terminal:

```
Send: M420 V
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     (  0,212)                                      (212,212)
Recv:         0       1       2       3       4       5       6
Recv:  6 | -0.431  -0.431  -0.264  -0.119  +0.039  +0.132  +0.225
Recv:    |
Recv:  5 | -0.431  -0.431  -0.264  -0.119  +0.039  +0.132  +0.225
Recv:    |
Recv:  4 | -0.361  -0.361  -0.217  -0.064  +0.067  +0.163  +0.259
Recv:    |
Recv:  3 | -0.279  -0.279  -0.127 [-0.011] +0.090  +0.172  +0.254
Recv:    |
Recv:  2 | -0.256  -0.256  -0.116  -0.013  +0.081  +0.138  +0.194
Recv:    |
Recv:  1 | -0.252  -0.252  -0.104  -0.020  +0.044  +0.083  +0.123
Recv:    |
Recv:  0 | -0.248  -0.248  -0.091  -0.020  +0.044  +0.083  +0.123
Recv:         0       1       2       3       4       5       6
Recv:     (  0,  0)                                      (212,  0)
Recv: 
Recv: Mesh is valid
Recv: Storage slot: 0
Recv: echo:Bed Leveling Off
Recv: echo:Fade Height 10.00
Recv: ok
```

mtwomey, 2019-04-18:
Now running Marlin 2 bugfix, same results. Using Chrome but also tested in Firefox (just in case it was a browser thing).

mtwomey, 2019-04-18:
Interesting... doing a little debugging - we can see that the numbers appear to be parsing correctly. It's strictly a visual issue in the tooltip:

[image]

jneilliii, 2019-04-18:
Yeah, I don't why your set-up would be any different than mine unless it was some operating system number formatting option or default browser font or something silly.  I'm just using the plotly javascript library to render the surface, so it definitely has to be on the rendering side given you see the data correctly in the stored data tab of the plugin's settings and obviously in console it's correct.

closed 2019-05-09 by stale[bot]