jneilliii/OctoPrint-BedLevelVisualizer #121: How do I setup the plugin?

NovaViper, 2019-04-18:
Hey, I'm trying to get the plugin to function with my printer connection established gcode script in Octoprint. The script looks like this:

`M190 S75
G28 ; home nozzle
G28 Z ; get more accurate Z axis 
G29 ; Generate mesh for BL-Touch
G1 X117.50 Y117.50 Z3.0 F4000.0 
M500 ; Save mesh to EEPROM
M117 Bed Leveling Mesh Complete...
M140 S0 ; turn off bed uncomment to turn off bed
M84 ; disable motors`

But it never seems to display the data in the plugin.. am I doing something wrong here?

jneilliii, 2019-04-18:
If you want it to automatically update with the above "printer connection" script you just need to add the `@BEDLEVELVISUALIZER` flag to instruct the plugin to collect data. 
```
M190 S75
G28 ; home nozzle
G28 Z ; get more accurate Z axis 
@BEDLEVELVISUALIZER
G29 ; Generate mesh for BL-Touch
G1 X117.50 Y117.50 Z3.0 F4000.0 
M500 ; Save mesh to EEPROM
M117 Bed Leveling Mesh Complete...
M140 S0 ; turn off bed uncomment to turn off bed
M84 ; disable motors
```

closed 2019-04-18 by jneilliii

NovaViper, 2019-04-21:
Thanks for the info! I got it working :D