jneilliii/OctoPrint-BedLevelVisualizer #123: Question: How can I correct the values?

NovaViper, 2019-04-27:
I notice that sometimes the values are skewed by alot (i.e. one corner would be 0.14+ and another would suddenly be 5.6+)
The graph looks like this:
[image]

And these are the values
[image]

jneilliii, 2019-04-27:
Based on the stored data, those huge variances are being reported by the firmware. But without seeing the matching terminal response from the leveling command it's hard to say.

NovaViper, 2019-04-27:
Here's the terminal:

`Recv: ok
Send: M155 S60
Recv: ok
Send: G29 T1
[...]
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 +5.337 -0.097 +0.077 +0.258 +0.467
Recv:  1 -0.212 -0.158 +0.003 +0.140 +0.240
Recv:  2 -0.195 -0.168 -0.040 +0.080 +0.192
Recv:  3 -0.105 -0.122 -0.005 +0.113 +0.192
Recv:  4 -0.155 -0.035 +0.010 +0.187 +0.217
Recv: 
Recv: X:232.00 Y:227.00 Z:11.08 E:0.00 Count X:18560 Y:18160 Z:4520
Recv: ok
Send: M140 S0
Recv: ok
Send: M155 S3
Recv: ok
Send: M500
Recv: echo:Settings Stored (635 bytes; crc 13652)
Recv: ok
Send: M280 P0 S90
Recv: ok`

jneilliii, 2019-04-27:
Yeah, it's definitely skewed data being returned by your firmware/probing process.  You can see from the first response line of points there is a `+5.337` being returned.

The plugin just displays graphically what the firmware reports, so you may want to look into why the firmware is reporting that way.  Is your probe hitting a physical obstruction like a bulldog clip or something?

closed 2019-04-27 by jneilliii

NovaViper, 2019-04-27:
The only thing I got it tape.. but this seems to happen every once in a while. I've even updated the firmware and modified it multiple times and it still occurs, though it occurs randomly at different points

NovaViper, 2019-04-27:
But I do notice that sometimes the probe does this 'double flick' behavior where it tries to come down but then it suddenly starts flicking the probe out twice before it ever touches. From what I was told before, it's something with the v3 probes and Creality boards (the C7 capacitor on the Z axis slot) causing misreads

jneilliii, 2019-04-27:
Which firmware are you using?

NovaViper, 2019-04-27:
Sorry for the late response. I'm using the bugfix for Marlin 2.0

NovaViper, 2019-04-29:
I went and ran debug mode when doing G29T and here's what the logs look like:

https://hastebin.com/newanigamo.makefile

I also noticed why at that one particular corner I had the really high value because I had tape on the corner that was actually sticking up enough to end up triggering the sensor way too early.