jneilliii/OctoPrint-BedLevelVisualizer #124: Failure to install on Raspbi Zero W...

WriteCodeEveryday, 2019-05-10:
The setup just times out and locks up the raspberry bi hard...

Requires a full reboot to get octoprint to come back :(

jneilliii, 2019-05-11:
Pi Zero W takes forever to compile Numpy.  You can try doing this ahead of time by running the following commands in a terminal session.

```
cd ~/oprint/bin/
./pip install numpy
```

Once that is completed and successful the plugin should install without issue.

closed 2019-05-11 by jneilliii

Kiron21, 2019-05-15:
If you have only 512MB of RAM you can try my solution:

stop Octoprint: /etc/init.d/octoprint stop
stop other services like mjpeg-streamer the same way.
instal numpy (example): /home/octoprint/OctoPrint/venv/bin/python -m pip install numpy -v