jneilliii/OctoPrint-BedLevelVisualizer #126: The settings are not properly documented

timur-tabi, 2019-05-11:
The pop-up help window that appears over each settings option just repeats what the setting is.  The README needs to have detailed documentation on each setting.  For example, I have no idea what "Bilinear Mesh Leveling" means, or whether I should turn it on, or even if it applies to my printer.

jneilliii, 2019-05-11:
Proper is a relative term.

timur-tabi, 2019-05-11:
Many of the options are not documented at all.

LMS0815, 2019-11-19:
Consider asking Dr. Google:

http://marlinfw.org/docs/features/unified_bed_leveling.html

jneilliii, 2020-03-25:
With the latest release all the options have been enhanced with better pop-over descriptions, closing issue for now. Will work on better documentation on the readme in the future.

closed 2020-03-25 by jneilliii