jneilliii/OctoPrint-BedLevelVisualizer #127: A wiki of options for known printers would be very helpful

timur-tabi, 2019-05-11:
If you look through the closed issues, you can get some hints as to how to configure this plugin for your specific printer, but that shouldn't be necessary.  Instead, there should be a wiki, or some other user-contributable document, that contains settings for various printers.

jneilliii, 2019-05-11:
I don't have time, I believe the wiki is public, you want to help?

closed 2019-05-11 by jneilliii

timur-tabi, 2019-05-11:
I think you need to enable the Wiki, because I can't access it.

jneilliii, 2019-05-11:
Does it show up at all?

timur-tabi, 2019-05-11:
I see on the top of the page where it shows "Wiki", but if I click on it, all it does is bring me to the main page.  So https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/wiki just redirects to https://github.com/jneilliii/OctoPrint-BedLevelVisualizer.

jneilliii, 2019-05-11:
Thanks, had to create the first page I think.  I added a page specifically for examples and put up the first one.

timur-tabi, 2019-05-13:
Thanks.  I was thinking about augmenting it with settings as well.  Is that okay?

jneilliii, 2019-05-13:
Sure.