jneilliii/OctoPrint-BedLevelVisualizer #128: Coordinates do not match

gandy92, 2019-05-15:
On a cartesian machine running Marlin bugfix-1.1.x with UBL, I get the following topology report:
```
Send: G29 T
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     ( 10,294)                                      (294,294)
Recv:         0       1       2       3       4       5       6
Recv:  6 | -0.126  -0.290  -0.440  -0.489  -0.497  -0.450  -0.292
Recv:    |
Recv:  5 | -0.193  -0.364  -0.513  -0.534  -0.563  -0.511  -0.355
Recv:    |
Recv:  4 | -0.161  -0.317  -0.439 [-0.484] -0.531  -0.465  -0.297
Recv:    |
Recv:  3 | -0.131  -0.310  -0.444  -0.498  -0.497  -0.444  -0.293
Recv:    |
Recv:  2 | -0.072  -0.268  -0.391  -0.466  -0.521  -0.439  -0.281
Recv:    |
Recv:  1 | -0.032  -0.232  -0.402  -0.480  -0.511  -0.452  -0.255
Recv:    |
Recv:  0 |   .       .       .       .       .       .       .
Recv:         0       1       2       3       4       5       6
Recv:     ( 10, 10)                                      (294, 10)
Recv: 
Recv: ok
```
The report includes the coordinates for the top/bottom/left/right corners. In this case, the MESH_INSET is set to 10. 

The Bed Visualizer, however uses different coordinates, which appear to be derived from the X/Y_POS_MIN/MAX values:
[image]

Would it be possible to scan the report for these coordinates in order to 
- get the coordinates right
- automatically choose x/y axis flipping (see #125)

jneilliii, 2019-05-15:
That's the issue. Every firmware and different versions of same firmware and different types of leveling reports the mesh differently. I took the approach that this is the easiest approach to support most. I know it's not accurate or perfect, but it's just supposed to be a representation to aid you in making bed adjustments.

gandy92, 2019-05-15:
The plugin is indeed a great help leveling the bed, if only to see at one glance that the topography data appears to be free of outliers. I do see the problem in having all those different output formats, however. Where the coordinate hints are missing, nothing could be done, anyway. If you agree it's worth a try I could attempt to find a solution for the case where the corner coordinates are provided and offer a PR.

jneilliii, 2019-05-15:
I'm more than happy to take PR requests, especially since I'm a self trained programmer. I'm sure there are better ways to do things than how I hack them together.

gandy92, 2019-05-20:
Hi, just submitted a hacked together (I love this phrase) PR with that functionality. I could only test it with my printer, so no guarantees it will work with others, though I tried to be careful.

commit b7c1325c (2019-05-26): 0.1.8

- Fix minor bug where lines w/ "1 |[+0.019] -0.137" are not matched
- Add grid boundary detection and auto flipping, resolves #128
commit 3c40fb5e (2019-05-26): 0.1.8

- Fix minor bug where lines w/ "1 |[+0.019] -0.137" are not matched
- Add grid boundary detection and auto flipping, resolves #128
commit a604b1dc (2019-05-26): 0.1.8

- Fix minor bug where lines w/ "1 |[+0.019] -0.137" are not matched
- Add grid boundary detection and auto flipping, resolves #128
closed 2019-05-26 by jneilliii