jneilliii/OctoPrint-BedLevelVisualizer #130: graph coordinates do not reflect measured coordinates

Sprk-nl, 2019-05-27:
First of all, this is a great tool, which I like to contribute to.
Before diving in, I would like to have some feedback.

The printer has a bed of 235x235. With probe offset I've defined the 
following X and Y min and max probe positions:
X: 40 - 195
Y: 40 - 195
Simply said: I have a margin of 40cm from each side.

The graph displays the first probed position being X40 Y40 as X0 Y0.
And following this, the probed position of X195 Y195 in graph is: X235 Y 235.

The bed width (X) and depth (Y) come from the printer's profile.
I want to draw the bed a different color in the graph, or as a transparant box with a fat line.
The measurements would be using the actual probed coordinates.

How would I start? (with on Klipper)

- Discover firmware:
As different firmwares and plugins could be used to probe, I'm thinking of a function that will auto-check which firmware (Marlin / Klipper) would be used. And being manual selectable.

- Read saved mesh data from klipper printer.cfg file. Could be multiple profiles.
Example:
```
#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [bed_mesh default]
#*# points =
#*# 	  0.254167, -0.065833, 0.276667
#*# 	  0.127500, -0.030000, 0.075000
#*# 	  -0.088333, 0.035000, 0.140000
#*# x_count = 3
#*# y_count = 3
#*# min_x = 80.0
#*# max_x = 235.0
#*# min_y = 49.0
#*# max_y = 204.0
#*# x_offset = -40.0
#*# y_offset = -9.0
#*# mesh_x_pps = 2
#*# mesh_y_pps = 2
#*# algo = lagrange
```

At Klipper boot I also see the probed coordinates passing by.
The information should be somewhere :) 
And that's what I'm after, if it fits the projects plan.

Klipper boot:
```
[bed_mesh default]
points =
        0.371667, -0.092500, 0.261667
        0.180000, -0.004167, 0.153333
        -0.021667, 0.050833, 0.151667
x_count = 3
y_count = 3
min_x = 80.0
max_x = 235.0
min_y = 49.0
max_y = 204.0
x_offset = -40.0
y_offset = -9.0
mesh_x_pps = 2
mesh_y_pps = 2
algo = lagrange
tension = 0.2
=======================
bed_mesh: generated points
0: (80.0, 49.0)
1: (157.5, 49.0)
2: (235.0, 49.0)
3: (235.0, 126.5)
4: (157.5, 126.5)
5: (80.0, 126.5)
6: (80.0, 204.0)
7: (157.5, 204.0)
8: (235.0, 204.0)

```

jneilliii, 2019-05-28:
The points were always derived based on the overall need dimensions until the last version with changes from @gandy92 that detects the offsets reported by marlin unified bed leveling. Depending on how klipper reports, similar techniques may be incorporated to derive those offsets.

jneilliii, 2019-05-28:
Looking at #92 you may be able to use the command `BED_MESH_MAP` and read the line that is returned that starts with `mesh_map_output` as it lists the min and max points in a json string along with all the z positions. The drawback in my opinion is that it does a whole probing process versus in Marlin UBL where you can just run a report of the current mesh.

jneilliii, 2022-01-30:
Added the option to define probe margin/offset in above commit and is available in version 1.1.2rc2. You can change the plugin's release channel in OctoPrint's Software Update settings to `Release Candidate` and update when prompted if you want to try it out.

closed 2022-01-30 by jneilliii