jneilliii/OctoPrint-BedLevelVisualizer #131: ABL_BILINEAR_SUBDIVISION definitely mangles the graph (affirmation)

filipgoc, 2019-05-31:
Hi, one, great idea and implementation with the Bed Visualizer and plot.ly! Kudos. 

I know this is a duplicate, but the other issues are closed and didn't seem to result in a clear consensus. 

Especially https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/89 seems to end in awash. 

I performed this test to make sure there's an issue and to give you sample data. 

---

## TL:DR
The abl_subdivision definitely throws the Bed Visualizer off. No doubt. 
The visualizer tries to treat both tables (the source values and the subdivided values) as one table, and that cannot work - they don't even have the same number of rows and columns, and everything gets messed up. 

I saved two graphs in plot.ly so you can have a look. 
I also looked at the 'stored data' in bed visualizer settings, and indeed it is the same mess (no surprise there.)

ABL_BILINEAR_SUBDIVISION OFF
https://plot.ly/~fffilip/13/
[image]

ABL_BILINEAR_SUBDIVISION ON
https://plot.ly/~fffilip/15/
[image]

## PS: Experimental Constants

I have bilinear leveling with 6 points in each direction.
For the two tests: 
Temperature was constant (+/- 1deg)
The printer was not touched by human hand during this test. 
The plate was tilted; I know, it makes for a nice graph. 
After each firmware flash, I ran

```
M503
M500
G28
G29
M420 V
```

jneilliii, 2019-06-01:
Thanks for the detailed description.  Hopefully you still have the bilinear option flashed to your printer.  Could you please post your gcode settings that you are using in the plugin, and the results from the terminal tab of the command that follows the @BEDLEVELVISUALIZER line?

filipgoc, 2019-06-01:
It's fast to flash it back :-)

My settings in the test above were
```
M155 S30
M420 V
M155 S3
```

Now based on what you said above I injected the @BEDLEVELVISUALIZER, but the results are the same. 
```
M155 S30
@BEDLEVELVISUALIZER
M420 V
M155 S3
```

Here's the terminal output (no @BEDLEVELVISUALIZER to be found, but it's the section where the data is requested...)

```
Send: M155 S30
Recv: ok P15 B3
Send: M420 V
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3
Recv:  0 +0.011 -0.050 -0.094 -0.188
Recv:  1 +0.037 +0.028 +0.017 -0.039
Recv:  2 +0.047 +0.080 +0.096 +0.055
Recv:  3 +0.111 +0.151 +0.162 +0.125
Recv: 
Recv: Subdivided with CATMULL ROM Leveling Grid:
Recv:         0        1        2        3        4        5        6        7        8        9
Recv:  0 +0.01125 -0.00997 -0.03119 -0.05037 -0.06425 -0.07609 -0.09363 -0.12133 -0.15473 -0.18813
Recv:  1 +0.02036 +0.00541 -0.00955 -0.02326 -0.03312 -0.04174 -0.05572 -0.07902 -0.10768 -0.13633
Recv:  2 +0.02947 +0.02079 +0.01210 +0.00386 -0.00199 -0.00740 -0.01781 -0.03672 -0.06063 -0.08454
Recv:  3 +0.03687 +0.03401 +0.03114 +0.02800 +0.02603 +0.02378 +0.01663 +0.00139 -0.01874 -0.03887
Recv:  4 +0.03945 +0.04207 +0.04468 +0.04644 +0.04854 +0.04979 +0.04575 +0.03319 +0.01534 -0.00251
Recv:  5 +0.04032 +0.04797 +0.05561 +0.06191 +0.06794 +0.07262 +0.07140 +0.06078 +0.04425 +0.02772
Recv:  6 +0.04713 +0.05857 +0.07001 +0.07963 +0.08827 +0.09508 +0.09562 +0.08634 +0.07080 +0.05525
Recv:  7 +0.06438 +0.07773 +0.09108 +0.10208 +0.11120 +0.11796 +0.11862 +0.10991 +0.09511 +0.08031
Recv:  8 +0.08756 +0.10160 +0.11563 +0.12679 +0.13506 +0.14045 +0.14019 +0.13145 +0.11705 +0.10265
Recv:  9 +0.11075 +0.12546 +0.14018 +0.15150 +0.15892 +0.16294 +0.16175 +0.15298 +0.13899 +0.12500
Recv: 
Recv: echo:Bed Leveling ON
Recv: ok P15 B3
Send: M155 S3
```

jneilliii, 2019-06-01:
What about replacing the M420 V line with G29 T?

filipgoc, 2019-06-01:
Thanks for looking into this. I wonder: why could it would make a difference? Does not the visualizer just grab the terminal output? The output of G29 T and M420 V is the same as far as I know. 

1.  I tested it, and it seems to output the same thing.
```
Send: M155 S30
Recv: ok P15 B3
Send: G29 T
[...]
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3
Recv:  0 -0.000 -0.055 -0.087 -0.174
Recv:  1 +0.020 +0.037 +0.027 -0.027
Recv:  2 +0.036 +0.086 +0.111 +0.079
Recv:  3 +0.091 +0.156 +0.183 +0.162
Recv: 
Recv: Subdivided with CATMULL ROM Leveling Grid:
Recv:         0        1        2        3        4        5        6        7        8        9
Recv:  0 -0.00013 -0.01928 -0.03843 -0.05500 -0.06520 -0.07281 -0.08663 -0.11162 -0.14281 -0.17400
Recv:  1 +0.00680 -0.00316 -0.01312 -0.02258 -0.02931 -0.03554 -0.04746 -0.06905 -0.09634 -0.12363
Recv:  2 +0.01373 +0.01296 +0.01219 +0.00983 +0.00658 +0.00174 -0.00830 -0.02649 -0.04988 -0.07327
Recv:  3 +0.02013 +0.02688 +0.03363 +0.03737 +0.03776 +0.03513 +0.02750 +0.01254 -0.00742 -0.02737
Recv:  4 +0.02417 +0.03588 +0.04759 +0.05600 +0.06070 +0.06211 +0.05814 +0.04632 +0.02913 +0.01194
Recv:  5 +0.02769 +0.04268 +0.05768 +0.06974 +0.07893 +0.08520 +0.08543 +0.07657 +0.06167 +0.04677
Recv:  6 +0.03562 +0.05329 +0.07096 +0.08587 +0.09835 +0.10808 +0.11137 +0.10487 +0.09194 +0.07900
Recv:  7 +0.05110 +0.07097 +0.09084 +0.10762 +0.12131 +0.13190 +0.13622 +0.13112 +0.11974 +0.10837
Recv:  8 +0.07099 +0.09245 +0.11392 +0.13175 +0.14545 +0.15552 +0.15974 +0.15543 +0.14527 +0.13512
Recv:  9 +0.09087 +0.11394 +0.13700 +0.15587 +0.16959 +0.17913 +0.18325 +0.17974 +0.17081 +0.16188
Recv: 
Recv: echo:enqueueing "G0 Z10"
Recv: X:131.00 Y:111.00 Z:3.04 E:0.00 Count X:10480 Y:8880 Z:5071
Recv: ok P15 B2
Recv: echo:enqueueing "G0 X-8 Y20 F6500"
Send: M155 S3
```

2. There's also an issue with G29 T in that it performs bed leveling. I would prefer the Bed Visualizer to simply visualize the mesh I have saved in EEPROM, not generate and visualize a new mesh. I prefer to generate meshes on my own time and terms.

jneilliii, 2019-06-01:
Thanks, I just wanted ro confirm.

jneilliii, 2019-06-15:
I may have a fix for this issue now.  Please test by installing the plugin using the URL below in plugin manager and let me know how it goes.  

> There's also an issue with G29 T in that it performs bed leveling. I would prefer the Bed Visualizer to simply visualize the mesh I have saved in EEPROM, not generate and visualize a new mesh. I prefer to generate meshes on my own time and terms.

This is kind of handled by the plugin by storing the mesh that was last retrieved by pressing the Update button if the store mesh data option is enabled. But I see your point.  Does the `M420 V` command not run the leveling process?  If not, then you could use that in the plugin's settings and then run the leveling outside of the plugin's update button at any time and then read in the stored settings using the update button. Disabling the store mesh data option will run the commands every time you switch to the tab if I'm not mistaken, so it's like pressing the update button.

filipgoc, 2019-06-16:
I've tested it now. It works. You've fixed it. Good job!

---

Oh yeah, M420 V works great for me; it's what I had used before. I only tried G29 T as per your suggestion, and I was expressing why I prefer M420 V.

filipgoc, 2019-06-16:
I think this can be closed 👍

PR #134 (2019-06-16): 0.1.9
**Fixed**
- Issue where ABL_BILINEAR_SUBDIVISION in Marlin was not graphing correctly.  Now the mesh will be reset to an empty set if the phrase `Subdivided with CATMULL ROM Leveling Grid:` is received and continue mesh collection of points. Resolves #131.
closed 2019-06-16 by jneilliii