jneilliii/OctoPrint-BedLevelVisualizer #132: Not showing properly for DAGOMA Magis

hyryel, 2019-06-08:
Hi, 
this plugin look like cool, but it doesn't work with my printer.
It is a Dagoma Magis with a circular bed.

I've checked the option "Use center origin", but i got a messy graph.
[image]

Am I doing something wrong or circular bed are not supported ?

jneilliii, 2019-06-08:
What are the commands you set in the plugin settings and which firmware does your printer use?

hyryel, 2019-06-08:
Wow, quick answer.

here there are the commands :
M155 S0 //disable temperature info
G28 //auto home
@BEDLEVELVISUALIZER
G29 T //bed leveling
M155 S3 //enable temperature info

My printer uses a customized Marlin firmware from Dagoma.

I will see if I can get more info on it

jneilliii, 2019-06-08:
If you run G29 T on the terminal tab, what is returned? There's a good chance that the marlin version is real old and the breed report isn't recognized by the plugin properly.

hyryel, 2019-06-09:
I tried the G29 T command and here is what I receive :
[sample.txt](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/3269037/sample.txt)

jneilliii, 2019-06-12:
Yeah, it appears that the report coming out is reporting multiple lines for the same data and it's causing issues with the parser.  For example, the lines below show the duplicates for the same points, but is adding `Count`, which I've never seen before. I'd need more info on their firmware, or possible you could use a different verbosity level (if that firmware supports it) by using something like `G29 V0 T` or some other number other than 0. It would be a lot of trial and error.
```
X:0.00 Y:0.00 Z:227.16 E:388.82 Count X: 33201 Y:33201 Z:33201
Bed X: -81.29 Y: -46.93 Z: 0.28
X:-81.29 Y:-46.94 Z:0.27 E:388.82 Count X: 16822 Y:10655 Z:10655
Bed X: -81.29 Y: -46.93 Z: 0.28
```

hyryel, 2019-06-12:
My printer is this one [Dagoma Magis](https://dagoma.fr/boutique/produit/imprimantes-3d/imprimante-3d-neva-magis.html#block-technical-features).

The bed leveling procedure is as follow :
Reaching a position
touch between 3 and 5 time (perhaps to be sure to obtain the same value multiple time)
proceed to next position.

Do this for 12 positions.
I think this is why you see multiple lines for the same data.

jneilliii, 2019-06-17:
Oh, so it's kind of a guided manual leveling process, not an automatic bed leveling procedure? What do you get it you send the command `M420 V`?

hyryel, 2019-06-17:
In fact, the bed leveling is totally automated.
Here is the result of command M420 V :
Send: N161723 M420*21
Recv: ok

I don't know why Octoprint add "N161723" at the beggining and  "*21" at the end of command.
M420 V return nothing. That too bad.

jneilliii, 2019-06-17:
Yeah, I looked at the latest dogma firmware and it appears to be an old Marlin 1.1.0-RC6 variant.  I recently worked with someone else that was having issues with this old Marlin version and we ended up having to upgrade the firmware completely to get the required leveling data for the plugin to work properly.  You may want to search around to see if anyone has a newer fork of Marlin (latest stable is 1.19.0) and flash that.

closed 2019-07-08 by stale[bot]