jneilliii/OctoPrint-BedLevelVisualizer #133: [BUG] "First usage"(?) settings dialog appears EVERY refresh/load OctoPrint web page.

gruvin, 2019-06-10:
Hi

Every time I load or reload the OctoPrint web interface, Bed Visulizer's (presumably first usage "only") _Setup Wizard_ dialog appears. I have to click _Finish_ every time,  before I get to use OctoPrint.

I do have quite a few other plugins installed. That may be part of the cause but shouldn't be, of course.,

Thanks for your work. As a no probe manual mesh editor, this plugin is SUPER useful to me. 😄 

Screen cap of what I see, for clarity ...

[image]

jneilliii, 2019-06-10:
Haven't seen that issue before. Try opening settings after you get into the interface and go to the bed visualizer area and verify your settings and save there. Then it might not show next time.

gruvin, 2019-06-10:
I did that, actually. No change.

Can you let me know the mechanism / yaml entry or whatever it is that should be getting set to stop the first time dialog? I could set it manually, perhaps?

Thanks.

jneilliii, 2019-06-12:
The wizard hooks are tied to the following section in config.yaml
```
plugins:
  bedlevelvisualizer:
    command:
```
So based on the functions [here](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/a604b1dc3b1aeb8b21fd128428a256b62c8f1f4b/octoprint_bedlevelvisualizer/__init__.py#L50) if that is blank it will trigger the wizard to show.

gruvin, 2019-06-12:
Thanks for the link to the code in question, which made it crystal clear that this couldn't be a bug with bedlevelvisualizer code at all. It's a pretty concise test, after all. 

So, I figured it out in the end. It was the freakin browser cache. Unbelievable! Way to go Chrome. 😖 

Thanks for your time.

closed 2019-06-12 by gruvin

nielsjl, 2019-06-12:
Thanks gruvin. I was experiencing the exact same problem. Restarting Chrome fixed it. 👍