jneilliii/OctoPrint-BedLevelVisualizer #135: Timeout occured - where is the timeout setting

WolfgangFahl, 2019-06-18:
I get "... error timeout occured before ..." while the G29 T command is still in progress.
Where is the setting to increase the timeout?
Why does the timeout happen in the first place?

jneilliii, 2019-06-18:
Thanks for the call out on the typo. The timeout is there for instances when the process just never finishes and allow the interface to recover from processing. It's set at 60 seconds because the bed report can be retrieved without running a full probe process for some firmware/leveling types. For example you can use `M420 V` in marlin and just report the points.

closed 2019-07-09 by stale[bot]

nickm324, 2019-07-10:
Same issue, is this being looked into? Is the time out configurable in some way?

jneilliii, 2019-07-10:
Timeout is configurable in the plugin's settings.

WolfgangFahl, 2019-07-10:
The message disappears so quickly that it is not obvious. People start searching the internet and end up here.