jneilliii/OctoPrint-BedLevelVisualizer #136: Listen for G29 output

lostcontrol, 2019-06-20:
I trigger bed leveling most of the time from the printer lcd (Marlin firmware) or via Printoid's command line interface. BedLevelVisualizer seems to only update the mesh when G29 is trigger from the plugin. It would be great if the plugin could also update the visualization when G29 is triggered from somewhere else.

jneilliii, 2019-06-21:
I'm not sure if that would be possible. Too many different firmware variants, etc. You can however possibly change ther plugin's commands to use `M420 V` instead of G29 to just graph the stored eeprom values if you're using a newer version of marlin.

lostcontrol, 2019-06-21:
Yes, nice trick with `M420 V` :smiley: That works perfectly and fulfill my use case(s). Thank you.

closed 2019-06-21 by lostcontrol