jneilliii/OctoPrint-BedLevelVisualizer #137: feedrate

forthereg, 2019-06-21:
Thank you for the Plug, its fantastic.
the move nozzle in bed visualizer is a wonderful feature. But moves at about 1mm/second for me, anyway to change the feedrate on that to something usable ?    thank you.

jneilliii, 2019-06-21:
So the plugin only sends the command `G1 X<position> Y<position>` as a command so your default speeds of your firmware are what control that.

forthereg, 2019-06-23:
Riggghhhhtttt ... that makes so much sense , so its just using the last feedrate which is always the secondary accurate probing for me when ever Im using this plug in .
So i will just give it a move command in octoprint with a "normal" F before i use the Graph nozzle move . 
thank you .

jneilliii, 2019-06-23:
You could add that command to the end of the plugin's settings.

closed 2019-06-23 by jneilliii