jneilliii/OctoPrint-BedLevelVisualizer #138: Two issues, I think are related.

Kotawolf, 2019-06-22:
I have it run the code:
G28	
@BEDLEVELVISUALIZER	
G29 T	

At the end of the sequence I'm getting the following error:
TypeError: t._fullLayout._defs is undefined

CR-10s

The second issue, is that the picture it gives, is very narrow, it isn't taking up the full width of the tab.   So, something going on with the layout, but I can' find a layout option.   Trying to attach a screen shot.
[image]

Kotawolf, 2019-06-22:
Nevermind, dug into history and found the fix for the first one, and the second one may just be browser is being dumb.

closed 2019-06-22 by Kotawolf

jneilliii, 2019-06-22:
Could also be themeify...