jneilliii/OctoPrint-BedLevelVisualizer #139: Please wait, retrieving current mesh.  You must be logged in.

jkiejr, 2019-06-29:
Just recently updated to octoprint Version 1.3.11 and whenever i launch octoprint i get title message in bed visualizer. I believe i have put in the proper commands to level the bed, a G28 followed by a G29. I click update and it does nothing.
[image]

jneilliii, 2019-06-29:
Verify the settings are correct in the plugin's settings and save, I recommend you look at the wiki for examples...then make sure to do a [full browser refresh](https://en.wikipedia.org/wiki/Wikipedia:Bypass_your_cache).

jkiejr, 2019-06-29:
So not sure what settings need to be checked or not but this is what i have
[image]

jneilliii, 2019-06-29:
That will depend on your firmware...there are multiple examples [here](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/wiki/Firmware-Specific-GCODE-Examples).

jkiejr, 2019-06-29:
I tried it on Internet explorer and Edge, still same issue

jneilliii, 2019-06-30:
Make your GCODE commands the following, check the box to save mesh and restart octoprint and force a refresh on your browser.

``
M155 S30
@BEDLEVELVISUALIZER
G29 T
M155 S3
```

jkiejr, 2019-06-30:
it doesn't appear to be saving anything when i restart octoprint, even after i hit save in the settings

jneilliii, 2019-06-30:
Then there could be a javascript error.  If you open your developers console (F12 key on your keyboard typically) to see any errors on initial page load?

jkiejr, 2019-06-30:
There is a connection issue
[image]

jneilliii, 2019-06-30:
Need you to click on the packed_plugins part on the first error. Also, upload your octoprint.log.

jkiejr, 2019-06-30:
It appears it might have been an issue with octoprint, just installed new update today and it's fixed

closed 2019-06-30 by jneilliii