jneilliii/OctoPrint-BedLevelVisualizer #140: GCode for bed heating ignored

benebrady, 2019-07-05:
When issuing the command to generate the mesh, I'm requesting the bed to be heated to 65 degrees C... but the Bed Visualizer Plugin seems to ignore the gcode.

Here is the code I'm currently trying to use. No bed heating occurs...
M190 65 ;heat bed and wait for temperature
M155 S30
G28 ; home the printer
@BEDLEVELVISUALIZER
G29 T ; Generate the mesh
M155 S3
M140 S0 ; Turn off the bed

jneilliii, 2019-07-05:
The plugin just sends the commands to the printer so not sure why this wouldn't work. I'll test tonight and see if I'm able to reproduce.

LMS0815, 2019-08-29:
> When issuing the command to generate the mesh, I'm requesting the bed to be heated to 65 degrees C... but the Bed Visualizer Plugin seems to ignore the gcode.
>
> Here is the code I'm currently trying [...]

Try M190 S65 

see: http://marlinfw.org/docs/gcode/M190.html

jneilliii, 2020-03-25:
Assuming this is good now since there's been no activity. Closing, feel free to comment if @LMS0815 recommendation didn't fix the issue for you.

closed 2020-03-25 by jneilliii