jneilliii/OctoPrint-BedLevelVisualizer #141: Installation Hangs with "Running setup.py install for numpy: started"

xenon2008, 2019-07-06:
Hello together!

I want to install Bed Visualizer for the first time on my Octopi (latest Version), but the installation always hangs on "Running setup.py install for numpy: started"

What can i do?
Tried it 2-3, but every try the same error ... i must reboot Octopi via SSH so that was working again...

Sorry for my Bend English, i am a German User with terrible English Skills

jneilliii, 2019-07-06:
Check this post and see if this works for you for installing the numpy requirement.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/81#issuecomment-421539386

xenon2008, 2019-07-07:
is there no other "official" solution?

jneilliii, 2019-07-07:
If you can't get numpy to install then the newer version won't work for you. The only other option would be to install the last version prior to the numpy integration which doesn't support all firmware. Once installed you can't upgrade otherwise you'll get the same error.  To install the old version use the following url in plugin manager.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.2.zip

closed 2019-07-28 by stale[bot]

saikek, 2019-10-15:
Issue happens on low memory machines. 

## Installation on machine with memory < 512 MB.
If Your Octoprint is installed on machine with 256 MB of ram ( for example Quad Core Orange PI Zero with 256MB RAM ) - you installation could fail during `numpy` compilation.  Your memory and swap would be 100% full as well as all CPU's. One of the solution is to temporary increase swap size for the installation:

```
sudo fallocate -l 1G /swapfile
sudo chmod 600 /swapfile
sudo mkswap /swapfile
sudo swapon /swapfile
```

this would create 1GB temporary swap file that can be used during installation. You can remove it after reboot.

jneilliii, 2019-10-15:
Can you include instructions on how to remove the swapfile prior to me accepting the pull request?

saikek, 2019-10-16:
Sure:
# Disabling Swap
```
swapoff -v /swapfile
rm /swapfile
```

gasrovescio, 2020-01-30:
pi@octopi:~ $ swapoff -v /swapfile
swapoff /swapfile
swapoff: Not superuser.
pi@octopi:~ $ rm /swapfile
rm: remove write-protected regular file '/swapfile'? s
pi@octopi:~ $ swapoff -v /swapfile
swapoff /swapfile
swapoff: Not superuser.

  WHY ?

jneilliii, 2020-01-30:
Maybe you need to sudo those commands? I'm not sure because I've never used a low memory device. 

```
sudo swapoff -v /swapfile
sudo rm /swapfile
```

gasrovescio, 2020-01-30:
to be more precise I use the Putty program to access Raspberry, then enter name (PI) and password (raspberry), then copy and paste these two command lines and I press enter, but writes that I am not a superuser.

saikek, 2020-01-30:
Yes, you need root permissions for that, use `sudo` before command.

gasrovescio, 2020-01-30:
sorry but I don't understand, after logging in I copy and paste the command ( sudo swapoff -v /swapfile
sudo rm /swapfile ) directly and press enter, what am I wrong?

saikek, 2020-01-30:
Depending on what you want to do - disable swap after installation ?  (since this 2 commands are meant to disable extra swap only).

You login to machine, paste command one by one and press enter after each one so it would be consumed by mighty ssh server.

rgauny, 2020-11-16:
Shouldnt this use piwheels?

jneilliii, 2020-11-16:
It does in a python 3 environment, but piwheels doesn't have python2 version compiled.

hyreto, 2020-12-12:
Hi, it stopped always at the same state.  anyone a solution? i have a raspi 3B

Stored in directory: /tmp/pip-ephem-wheel-cache-M7k4dA/wheels/42/13/55/6c52a13f5f081fda60bf4ef6dfb543143e9b07062badc41152
Building wheel for numpy (setup.py): started
Building wheel for numpy (setup.py): still running...
Building wheel for numpy (setup.py): still running...
Building wheel for numpy (setup.py): still running...
Building wheel for numpy (setup.py): still running...
Building wheel for numpy (setup.py): still running...

jneilliii, 2020-12-12:
Are you installing on a pi 3b or pi 4? I recommend upgrading to python 3 using the script by @cp2004 then run `sudo apt install libatlas3-base` and then install the plugin. That way numpy doesn't have to compile.

skopy123, 2020-12-28:
installer is still broken. I am running octoprint o raspberry pi 3A. I updated to latest version of octoprint (1.5.2 stable) and instalation get stuck on "Building wheel for numpy (setup.py): still running...". I never touch swap settings etc. I am not much familiar with python and I have no idea what is piwheels, please consider to modify install script of your plugin to sort out this hasle and allow instalation without problem.

cp2004, 2020-12-28:
How long has it been stuck for? The installation is not *designed* to not work...

It can take over 1/2 an hour, maybe longer. On Python 3, the process is near instant since there is a precompiled wheel available.

jneilliii, 2020-12-28:
The 3A's memory footprint is the issue here. There are some users that have reported upping their memory block helped to get it installed, but I personally would upgrade your octoprint instance to python 3 and the install doesn't have to compile numpy then. Use the instructions [here](https://github.com/cp2004/Octoprint-Upgrade-To-Py3) to upgrade. If you want to you can try the steps to up the memory with these steps.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/141#issuecomment-542227338

Joonch, 2021-01-11:
Tip: Don't use `python3 -v` to check the version. Press the settings wrench, press system info, and scroll down to `env.python.version:`. I thought I was screwed because python reported that it was on version 3 through PuTTY because @cp2004 said the numpy wheel installation would be instant if you were on python version 3 since python 3 already has a wheel available. But upon checking, the _actual_ Octopi environment version was 2.7.16. That was why it was taking so long. So I had to wait about 15 minutes for it to compile but it eventually finished.  
Info: 
`browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.141 Safari/537.36
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1200
env.hardware.ram: 917016576
env.os.bits: 32
env.os.id: linux
env.os.platform: linux2
env.plugins.pi_support.model: Raspberry Pi 3 Model B Rev 1.2
env.plugins.pi_support.octopi_version: 0.17.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 19.3.1
env.python.version: 2.7.16
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.2
printer.firmware: Marlin Ver 1.70.2 BL (Creality3D)
`

HamidMandi, 2021-01-14:
> Issue happens on low memory machines.
> 
> ## Installation on machine with memory < 512 MB.
> If Your Octoprint is installed on machine with 256 MB of ram ( for example Quad Core Orange PI Zero with 256MB RAM ) - you installation could fail during `numpy` compilation. Your memory and swap would be 100% full as well as all CPU's. One of the solution is to temporary increase swap size for the installation:
> 
> ## Adding extra swap file:
> ```shell
> sudo fallocate -l 1G /swapfile
> sudo chmod 600 /swapfile
> sudo mkswap /swapfile
> sudo swapon /swapfile
> ```
> 
> this would create 1GB temporary swap file that can be used during installation. You can remove it after reboot.
> 
> ## Removing extra swap:
> ```shell
> sudo swapoff -v /swapfile
> sudo rm /swapfile
> ```

It works for me perfectly. I installed NumPy on an Allwinner A20. It takes about a half-hour but finally done!

D4koon, 2022-12-29:
A possible solution could be to use piwheels

pip3 install numpy -i https://www.piwheels.org/simple

jneilliii, 2022-12-29:
piwheels by default is already added to the available wheels, so it should find it, but that's a moot point since the numpy dependency was removed in newer versions.