jneilliii/OctoPrint-BedLevelVisualizer #142: Add output in degrees

hprotzek, 2019-07-15:
Would be very helpful to add the output in degrees for the bed leveling M3 screws, like in this sheet: https://docs.google.com/spreadsheets/d/1AC3NHQXZbEj1rc0tZtuaRyycx2clcZNnR04FSibEA04/edit?usp=sharing

jneilliii, 2019-07-15:
Interesting idea, but not practical in the world of multiple 3d printer suppliers I don't think.  Not everyone uses M3 screws to attached the bed or have 6 screws to tighten, etc..  I'm sure there is some math involved in calculating that somehow, but I'm already slammed with supporting so many different firmware variants, I don't think adding this in would be doable any time soon.  Would be completely open to accepting a pull request though if you were to develop the changes.

GreyArea1966, 2019-08-20:
I can't write code...but I've got an Excel sheet that will do this if hprotzek wants a copy...let me know.

It's VERY simple.
1. Assumes only four corners - both for readings and adjustment - great for my glass plate, maybe not for everyone else.
2. If takes the average height.
3. It takes whichever of the four readings are the furthest from that average
4. Works out if it's lower or higher
5. Calculates how many turns and in what direction to change that pin's height to the average.

It doesn't allow for any effect that there might be from a warped base, or the fact that on a rigid plane, you can't really move one point independently of the others (so chances are, if I wind down one corner, the opposite diagonal corner may go up by the same amount - or less, depending what your bed is made of.

So very much "buyer beware"...but it was fun to work out.

Or if you just want a quick and dirty rule - a full turn of an M3 bolt is 0.5mm linear travel. Which is huge, for our purposes...

LMS0815, 2019-08-29:
The math is no witch work.

Huge:
M3 - 0.5
M4 - 0.7
M5 - 0.8
M6 - 1.0

-> 360° / M3 huge * high = degrees to turn
= 360 °/ M3 huge * 2mm
= 360 °/ 0.5mm * 2mm =1440° = 4 turns

even easier if you would just evaluate full turns:
2mm / M3 huge = 2mm / 0.5 = 4 turns
2mm  / M4 huge ) 2mm / 7 = 2.86 turns

and multiply with 360°

closed 2020-03-28 by jneilliii

LMS0815, 2020-03-31:
Hi @jneilliii thank you for merging. Seems you missed my most advanced feature the "Screw visualization".
[image]

Was this by purposes?

jneilliii, 2020-03-31:
Yes, it was inconsistent and didn't work on some numbers for some reason. It's there, just hidden for now until I had time to debug.

PR #210 (2020-03-28): 0.1.13
Made webcam image not load when that option isn't enabled and only while processing, resolves #208 
Screw based adjustment degrees, thanks to @LMS0815, resolves #142