jneilliii/OctoPrint-BedLevelVisualizer #144: Please wait, retrieving current mesh. You must be logged in.

Stomer48, 2019-08-11:
Plugin worked fine for two weeks. Experience same like "jkiejr" as explained in issue of June 29th. Re-downloaded OctoPrint-SideBarTempGraph 1.3.11 as updated version, but to no avail. G Code is G28 followed by G29 T.
Uninstalled Bed leveling Visualizer and installed again. No difference.

Stomer48, 2019-08-11:
Sorry "Re-downloaded OctoPrint-SideBarTempGraph 1.3.11", must read: "OctoPrint 1.3.11 release:.

jneilliii, 2019-08-11:
There was a post on the OctoPrint issues tracker related to the wizard plugin causing your error over [here](https://github.com/foosel/OctoPrint/issues/3211).  Should be fixed in next version of OctoPrint, but seems like you can force a reload of the web interface 

> Until then, a forced refresh (holding shift & clicking on the reload button) should solve this.

closed 2019-08-11 by jneilliii

Stomer48, 2019-08-11:
Thank you for your prompt reaction! This is most helpful.

Best Regards,

Wim van Breen