jneilliii/OctoPrint-BedLevelVisualizer #145: Not Mesh Data Output (Tronxy x5sa [400])

ChrisVonFrosta, 2019-08-13:
G29 start autoleveling on machine.
After finish, no error, no output of mesh data.
G29 + parameter result in error:

`
Recv: Resend:64
Recv: //last:63 curr:65 rcv:N65 M105*20
Send: N64 G29 T*84
Recv:
Recv: //callback_pf error
Recv: //sd cmd error cnt:50839 len:13 :N64 G29 T*84
Recv:
Recv: ok
`

ChrisVonFrosta, 2019-08-13:
M420 V result in same error/output. 
the pid calibration ( M303 E0 S200 C8) do output.

jneilliii, 2019-08-13:
What do you have in the plugin's settings? What firmware and version?

ChrisVonFrosta, 2019-08-14:
Currently nothing. But i tried "all":
G29 or G29 T or M420 or M420 V.

Currently to check which command give mesh data output, i use the terminal.
I think if i find the a command on which one my printer spitt the data out, i could use it in the plugin.

jneilliii, 2019-08-14:
You are right, the plugin will only work if you figure out the command necessary to spit out the offset data.  Without that report there's nothing I can do to resolve this.  The firmware on your printer may just not have that option available.

closed 2019-09-04 by stale[bot]