jneilliii/OctoPrint-BedLevelVisualizer #147: Not ignoring "Bed Level Correction Matrix" block

AlbatrosHurleur, 2019-08-18:
Hello,

My printer's Marlin G29 reports always ends with a "Bed Level Correction Matrix" (I tried several combination of parameters and was not able to suppress it). This matrix is then added to "stored data", thus is being plotted.

[image]
[image]

Would it be possible to ignore this matrix to support my printer?

G29 P7 T output:
```
Recv: Eqn coefficients: a: 0.00011618 b: 0.00014890 d: 1.50255799
Recv: Recv: Bed Height Topography:
Recv:    +--- BACK --+
Recv:    |           |
Recv:  L |    (+)    | R
Recv:  E |           | I
Recv:  F | (-) N (+) | G
Recv:  T |           | H
Recv:    |    (-)    | T
Recv:    |           |
Recv:    O-- FRONT --+
Recv:  (0,0)
Recv:  -0.02505 -0.01255 -0.00380 +0.00245 +0.00620 +0.00995 +0.01745
Recv:  -0.01005 -0.00380 +0.00495 +0.01370 +0.01870 +0.01995 +0.02870
Recv:  -0.00130 +0.00370 +0.00120 +0.00120 -0.00130 +0.00120 -0.00255
Recv:  +0.01620 +0.01745 +0.01620 +0.01495 +0.00620 +0.00620 +0.01370
Recv:  +0.00995 +0.00870 +0.00870 +0.00745 +0.00620 +0.00995 +0.01745
Recv:  -0.01380 -0.01630 -0.01380 -0.01380 -0.01130 -0.00880 +0.00245
Recv:  -0.03255 -0.03130 -0.02880 -0.02255 -0.01880 -0.01255 -0.00630
Recv: 
Recv: 
Recv: 
Recv: Bed Level Correction Matrix:
Recv: +1.000000 +0.000000 +0.000116
Recv: -0.000000 +1.000000 +0.000149
Recv: -0.000116 -0.000149 +1.000000
```

Thanks a lots!

jneilliii, 2019-08-19:
Do you happen to know which version of Marlin and which type of leveling your printer is using?  Have you tried using the `M420 V` command instead or `G29 T1`?

AlbatrosHurleur, 2019-08-20:
The printer sends Marlin 1.1.0-RC6 at startup, I know the manufacturer has made customizations, I can ask if they changed something on bed leveling. I think it's using linear leveling.

I had not tried M420 V; but it seems it's not supported, nothing is displayed:
```
Send: N56 M420*22
Recv: ok
```

jneilliii, 2019-08-20:
I'll revisit.  I thought that the code already stopped gathering points once it received that line `Bed Level Correction Matrix:` but I'll check again tonight when I get home from work.

AlbatrosHurleur, 2019-08-23:
That would be great, we're several owning the same brand to be interested in using Bed Level Visualizer.

jneilliii, 2019-08-23:
Try this version and let me know how it works for you.  Just use the following URL in plugin manager.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.10.zip

AlbatrosHurleur, 2019-08-23:
It's working great, many thanks! 

[image]
[image]
[image]

I'll notify the other people I know that were interested in this option. Nice plugin!

jneilliii, 2019-08-23:
Thanks for testing.  I've merged the changes to the master branch and release an official 0.1.10 version with the changes.  They should be able to install directly from the plugin manager list.

closed 2019-08-23 by jneilliii