jneilliii/OctoPrint-BedLevelVisualizer #148: Enhancements

GreyArea1966, 2019-08-20:
Instead of "Obtaining mesh data please wait", could the plugin display the items in the list as it reads them?

Why? Because it's difficult sometimes knowing exactly how the visualiser has sequenced the points. I'm used to seeing graphs with X=0 and y=0 at bottom left, but others are used to it it at top left. Sometimes, the visualiser will read the points in a different order; does this mean it has changed how they display? It would be helpful if during the read we could literally SEE which point (x,y and when read, z) is being recorded (and maybe add a "sequence number" for reading order?)

So a complete list as it is being built would show;

Obtaining current mesh;

Sequence                 X                     Y                     Z
001                          0                      0                   0.24
002                          0                      50                 0.23
003                          0                      100               Reading...

etc until complete.

Also...could the system remember the previous reading and show two graphs as comparison?

I've done some work dropping the numbers into Excel and written a small routine that advises the best correction, but it's only based on the assumption of four readings at the corners and identifying which corner is furthest from the average bed height, and turning it accordingly (allowing for a nominal 360 degrees = 0.5mm linear travel, the standard for an M3 bolt). I'd be happy to provide the spreadsheet if anyone wants to tinker around!

jneilliii, 2019-08-20:
To be honest I don't think this makes sense because the plugin doesn't read in the offsets as the probing is happening, it reads in the full report at the end of the process instead.

GreyArea1966, 2019-08-20:
Ah I see. In that case can you just reassure me....my Printer used to read 9 points, and it would make an "S" shape, from 0,0 (bottom left) up to 200,200 (top right).

As I've now moved to a glass plate, I see no point in having nine points - so I reduced it to four. It now reads in a "C" shape, and thus the first reading is (to the eye) bottom right or 0,200.

When the data comes out, can I be sure that it has resequenced it correctly? I ask because at the end of the test it's also started throwing my bed in the opposite direction (to y+200 instead of y=0).

jneilliii, 2019-08-20:
The plugin doesn't change anything in regards to direction, it literally just reads in the points from the bed topography report.  So I'm not sure what has happened for you in switching from 9 to 4 points, but possibly your firmware adjustments made some change to the absolute/relative positioning?

GreyArea1966, 2019-08-20:
So to be clear, bed levelling's idea of 0,0 will always match the movements I see on the LCD screen of the printer?

Does that mean if I set origin to the centre of the plate, the four point grid would change from x = 0,200 and y = 0,200 to x = -100,+100 and y = -100,+100?

Should bed levelling be run in absolute or relative? And if it's one or the other, why doesnt' the routine put it in the right mode to start with? :-)

jneilliii, 2019-08-20:
All dependent on firmware. For example on delta printers 0,0 iis the center of a circle.

closed 2019-09-10 by stale[bot]