jneilliii/OctoPrint-BedLevelVisualizer #149: And one teensy tiny problem

GreyArea1966, 2019-08-20:
My visualiser used to home properly after the mesh was gathered. Now it sends the Y axis in the wrong direction - shoving the bed to the front of the machine rather than the back.

As far as I can tell, everything else works correctly...the Y axis isn't inverted or anything.

Any suggestions?

jneilliii, 2019-08-20:
What do you have in the plugin's settings for gcode commands?

GreyArea1966, 2019-08-20:
This;

G28
G29
M420 V

jneilliii, 2019-08-20:
What is returned when you type M420V in terminal tab?

GreyArea1966, 2019-08-21:
This...

Recv: Bilinear Leveling Grid:
Recv:       0      1
Recv:  0 -0.112 -0.192
Recv:  1 -0.080 -0.147
Recv:  2 -0.215 -0.102
Recv: 
Recv: echo:Bed Leveling On
Recv: echo:Fade Height Off
Recv: ok

jneilliii, 2019-08-21:
Nothing should be effecting the Y direction, but from the output it seems that you are only getting 6 probing points.  That's atypical, usually it's 4, 9, 16, 25, etc. based on the number of probe points defined in firmware, at least from what I've seen with other's examples and my own experience.  Either way, you should see if setting your settings to the following makes any difference.

```
G28
G29
@BEDLEVELVISUALIZER
M420 V
```

GreyArea1966, 2019-08-21:
Sorry, the 6 was a deliberate and recent test to ensure that at least the axes were what I was expecting (and they are). I'll try your suggestion and see what happens. Thanks for your time by the way...I do appreciate it. Sent from my Huawei phone-------- Original Message --------Subject: Re: [jneilliii/OctoPrint-BedLevelVisualizer] And one teensy tiny problem (#149)From: jneilliii To: jneilliii/OctoPrint-BedLevelVisualizer CC: GreyArea1966 ,Author Nothing should be effecting the Y direction, but from the output it seems that you are only getting 6 probing points.  That's atypical, usually it's 4, 9, 16, 25, etc. based on the number of probe points defined in firmware, at least from what I've seen with other's examples and my own experience.  Either way, you should see if setting your settings to the following makes any difference.
G28
G29
@BEDLEVELVISUALIZER
M420 V

GreyArea1966, 2019-08-21:
Didn't make any difference....bed is still stubbornly heading to y=220 instead of y=0 after bed level...but homes correctly to x=0 y=0 when I hit the home command in Octoprint.

Tried "Flip Y axis" in the bed leveller settings but that just seems to flip the axis on the graph.

jneilliii, 2019-08-21:
Yes,  extremely strange. As I mentioned the plugin itself doesn't change settings on the printer itself so the issue I think is more firmware related.

closed 2019-09-11 by stale[bot]