jneilliii/OctoPrint-BedLevelVisualizer #152: Reset Camera to defaults - Customizing

LMS0815, 2019-08-29:
The _Reset Camera to defaults_ button seems to reset the view to the current maximum.
e.g. 300/300 if the bed is 300x300
[image]

I would prefer a view, which is more or less the same as the default view which is shown when you open the tab. This would ease fine tuning and prevent me adjusting the wrong screw.
[image]

jneilliii, 2019-08-29:
Thanks for the detailed report.  I'll see what needs to be done to get plotly's default Reset Camera to defaults button to not mess with the orientation.

LMS0815, 2020-03-18:
Removed HOME button in push to avoid confusion.

closed 2020-03-18 by LMS0815