jneilliii/OctoPrint-BedLevelVisualizer #154: Cache current mesh, but not in config.yaml

Billiam, 2019-09-02:
Caching changing data (like the current mesh) in the config forces clients to perform a full octoprint refresh when they wake, since configuration data has changed.

Similar issue here: 
https://github.com/foosel/OctoPrint/issues/2950
https://github.com/fabianonline/OctoPrint-Telegram/issues/156#issuecomment-448173513

Disabling the save "fixes" the issue, but then of course requires a fresh mesh level to view current data.

LMS0815, 2019-11-19:
> Disabling the save "fixes" the issue, but then of course requires a fresh mesh level to view current data.

Which should be no issue, because you can read and update the mesh from EEPROM (Marlin UBL FW: G29 T) and trigger a refresh via the printer and/or Custom Control Editor Command(s).

jneilliii, 2020-08-01:
Thinking of adding a history database per @DodgeDeBoulet recommendation in above linked issue. I believe if the database is stored in the plugin's data folder it would get cleared on `uninstall and clean` option but will have to do additional testing on that.

DodgeDeBoulet, 2020-08-02:
> Thinking of adding a history database per @DodgeDeBoulet recommendation in above linked issue. I believe if the database is stored in the plugin's data folder it would get cleared on `uninstall and clean` option but will have to do additional testing on that.

One thing that concerns me about the plugin's data folder being cleared is the potential loss of that data when an uninstall/reinstall is required for troubleshooting an issue. I'm not familiar with OctoPrint's internals, but if there's a way to accommodate an option to backup/restore the history data, that would be a potential solution.

jneilliii, 2020-08-02:
All that data gets backed up with the built in backup/restore functionality and I believe only if you use the clean option during uninstall does that matter.