jneilliii/OctoPrint-BedLevelVisualizer #155: Feature Request: Use/capture bed leveling events not triggered by the plugin

faithhack, 2019-09-05:
It would useful if BLV could watch terminal output and use bedleveling data with the need to trigger the event or read the EEPROM.
For example, I usually do a G28 + G29 in my start GCODE which sets the bedleveling and out puts it to the terminal. It would be helpful to just switch to the Bed Level Visualizer tab and see the results with requesting an update during a active print.

jneilliii, 2019-09-05:
If you just add the @BEDLEVELVISUALIZER tag to your start gcode it will do this.

faithhack, 2019-09-05:
Awesome! That worked. The docs made it seem like @BEDLEVELVISUALIZER was for the settings in the plugin. Maybe update the docs to mention this is a feature for any gcode - unless I missed it and need a remedial reading class.

closed 2019-09-05 by faithhack