jneilliii/OctoPrint-BedLevelVisualizer #157: IndexError: index 1 is out of bounds for axis 0 with size 1

ctluciani, 2019-09-16:
Unlike #109 a reboot did not help me.

jneilliii, 2019-09-16:
They also upgraded their firmware...

ctluciani, 2019-09-17:
Thanks for the quick response!

Is there a minimum required Marlin version? I don't see one in the readme or anything. 

I know my firmware is old so I will update it. If that fixes the problem, I will submit a PR to catch this exception and at the very least log it to the console.

jneilliii, 2019-09-17:
I've tried accommodating as many firmwares as possible and as many versions.  Obviously this is  not always possible. I'm not sure what version of Marlin introduced the correct formatting that I'm currently able to parse. It would be helpful to know what is returned by the G29 T command.

closed 2019-10-08 by stale[bot]