jneilliii/OctoPrint-BedLevelVisualizer #160: TypeError: Cannot read property 'canvas' of undefined

Tomkee, 2019-10-03:
Hi,

just installed klipper and read there code you add to and everything work fine. Unfoftunatly I get error everytime 

TypeError: Cannot read property 'canvas' of undefined

My gcode commands
```

G29
G32
@BEDLEVELVISUALIZER
BED_MESH_OUTPUT
```
On Marlin everything was ok.

jneilliii, 2019-10-03:
Could you please share your logs and any errors shown in developer console?

Tomkee, 2019-10-04:
`Send: G29
Recv: // probe at 90.000,20.000 is z=0.472500
Recv: // probe at 162.500,20.000 is z=0.545000
Recv: // probe at 235.000,20.000 is z=0.607500
Recv: // probe at 235.000,127.500 is z=0.675000
Recv: // probe at 162.500,127.500 is z=0.672500
Recv: // probe at 90.000,127.500 is z=0.637500
Recv: // probe at 90.000,235.000 is z=0.727500
Recv: // probe at 162.500,235.000 is z=0.785000
Recv: // probe at 235.000,235.000 is z=0.807500
Recv: // Mesh Bed Leveling Complete
Recv: // Bed Mesh state has been saved to profile [default]
Recv: // for the current session.  The SAVE_CONFIG command will
Recv: // update the printer config file and restart the printer.
Recv: ok
Send: G32
Recv: // Unknown command:"G32"
Recv: ok
Send: BED_MESH_OUTPUT
Recv: // Mesh Leveling Probed Z positions:
Recv: // -0.187500 -0.115000 -0.052500
Recv: // -0.022500 0.012500 0.015000
Recv: // 0.067500 0.125000 0.147500
Recv: Mesh X,Y: 7,7
Recv: Search Height: 5
Recv: Mesh Average: 0.00
Recv: Mesh Range: min=-0.1875 max=0.1475
Recv: Interpolation Algorithm: lagrange
Recv: Measured points:
Recv:   0.067500  0.090556  0.109722  0.125000  0.136389  0.143889  0.147500
Recv:   0.045833  0.064321  0.078765  0.089167  0.095525  0.097840  0.096111
Recv:   0.015833  0.031728  0.043673  0.051667  0.055710  0.055802  0.051944
Recv:   -0.022500  -0.007222  0.004444  0.012500  0.016944  0.017778  0.015000
Recv:   -0.069167  -0.052531  -0.038920  -0.028333  -0.020772  -0.016235  -0.014722
Recv:   -0.124167  -0.104198  -0.086420  -0.070833  -0.057438  -0.046235  -0.037222
Recv:   -0.187500  -0.162222  -0.138056  -0.115000  -0.093056  -0.072222  -0.052500
Recv: 
Recv: ok`

jneilliii, 2019-10-04:
First off, take the G32 command out of your settings since it's unknown. That doesn't effect the plugin however and is not the cause of the error. I still need to see your developer console (F12) and also what browser are you using? There's a chance that your video drives may also not support webgl, I've seem that cause weird errors.

jneilliii, 2019-10-06:
Yeah, I took your output above and sent it through virtually and it's working fine with the plugin for me using Chrome browser.

[image]

lamberaw, 2019-10-12:
I'm also receiving the same error message. However, I hadn't reported it as a bug because it was only on Chrome for Android. The odd thing is when you check WebGL's website, you see the spinning box, so Chrome for Android support WebGL, just not fully?

Tomkee, 2019-10-12:
Ok so I am using Brave as my browser and have the same isue. When I chceck WebGL's i see spinning box (so I have support for WwbGL) but I don't see my bed visualision. On my chrom everything is working.

closed 2019-11-02 by stale[bot]

orangepeelbeef, 2019-11-30:
Confirm same issue in Brave but works in chrome

jneilliii, 2019-11-30:
Does Brave work on the link below and load the surface? 

https://plot.ly/javascript/3d-surface-plots/

orangepeelbeef, 2019-11-30:
On android brave works on that page just fine, on arch linux i get the same
webgl not supported.

Just stumbled across this, which sounds the same.
https://github.com/plotly/plotly.js/issues/3640

Confirm that with 'shields down'  or changing 'Cross-site device
recognition blocked' to 'all device recognition requests allowed' it works.

SpectreGadget, 2020-10-11:
I've started to get this error. Chrome on Windows 10. Webgl is supported and the link above (https://plot.ly/javascript/3d-surface-plots/) renders just fine.