jneilliii/OctoPrint-BedLevelVisualizer #164: Install failing on Pi 3 a+

TwoToneEddy, 2019-11-10:
Hi,
I don't appear to be able to get this to install, hangs on the numpy install on the gui but i'm not sure if that's the issue, here is the error in the log file:

2019-11-10 15:26:33,974 - octoprint.plugins.pluginmanager - INFO - Installing plugin from https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip
2019-11-10 15:34:02,852 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-11-10 15:36:07,408 - octoprint.plugins.pi_support - ERROR - Got an error while trying to fetch the current throttle state via "/usr/bin/vcgencmd get_throttled"
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pi_support/__init__.py", line 263, in _check_throttled_state
    state = get_vcgencmd_throttled_state(command)
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pi_support/__init__.py", line 137, in get_vcgencmd_throttled_state
    output = sarge.get_stdout(command)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/sarge/__init__.py", line 1503, in get_stdout
    p = capture_stdout(cmd, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/sarge/__init__.py", line 1473, in capture_stdout
    return run(cmd, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/fixes.py", line 24, in decorated_function
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/sarge/__init__.py", line 1462, in run
    p.run(input=input, async_=async_)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/fixes.py", line 24, in decorated_function
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/sarge/__init__.py", line 1071, in run
    self.run_node(node, input=input, async_=False)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/fixes.py", line 24, in decorated_function
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/sarge/__init__.py", line 1187, in run_node
    result = getattr(self, method)(node, input, async_)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/fixes.py", line 24, in decorated_function
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/sarge/__init__.py", line 1333, in run_command_node
    node.cmd.run(input=input, async_=async_)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/fixes.py", line 24, in decorated_function
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/sarge/__init__.py", line 662, in run
    self.stdin = p.stdin
UnboundLocalError: local variable 'p' referenced before assignment
2019-11-10 15:44:04,149 - octoprint.plugins.pluginmanager - ERROR - Installing the plugin from https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip failed, could not parse output from pip. See plugin_pluginmanager_console.log for generated output

jneilliii, 2019-11-10:
Have you tried pre-installing numpy?

LMS0815, 2019-11-19:
I can confirm it is running on a PI 3+

mikekscholz, 2019-12-02:
Im  just starting to investigate what the problem is, But I too am failing on building the wheel for numpy, but on a 3B+ with a clean install this morning of the current latest stable octopi image, I will update as I figure out the problem.

yusijs, 2019-12-10:
I reinstalled octoprint on my Pi 3b+ yesterday, and same issues there.

closed 2019-12-31 by stale[bot]

ethanmckellar, 2020-06-22:
I am stilll having this issue with 3a+ can't install numpy outside of octoprint either

jneilliii, 2020-06-22:
Yeah, this is an issue related to the small memory footprint of the Pi 3A devices. I did find a solution that someone had sent me somewhere else, linked below to add a USB drive as swap memory. You could try that, or others have reported installing inside a larger memory Pi and then swapping the card over works as well.

https://github.com/rezaxdi/tensorflow-on-orangepi-zero/blob/master/GUIDE.md#installing-usb-memory-as-swap

ethanmckellar, 2020-06-23:
Installed numpy, not sure what combination of things worked. Definitely didn't do it through octoprint, ended up leaving it overnight after using conda I believe? Works fine now. Also my printtimegenius now works flawlessly 😂