jneilliii/OctoPrint-BedLevelVisualizer #165: BedLevel Visualizer - Reset/Restart

Thisismydigitalself, 2019-11-11:
It takes few runs to level a bed.
if between runs BLTouch needs to be reset, this means one must refresh octoprint page in order to see the update button as when BLV is waiting for printer to start it only shows "Please wait, Retrieving current mesh" message and no way to do anything other than to reload the page.

Bottom line is: UPDATE button needs to be available at all times, or a REFRESH page button.

Thank you for a great tool!

jneilliii, 2019-11-11:
There is a timeout value that should stop that processing part and return the interface back to normal.

Thisismydigitalself, 2019-11-11:
Oh, now i see. it was set to 100sec..
I will check if 5sec will suffice.

closed 2019-12-02 by stale[bot]