jneilliii/OctoPrint-BedLevelVisualizer #166: Buttons for any User-defined macros

EagleB3, 2019-11-14:
Please offer on the plugin’s tab, somewhere close to the “Update” button, 
1..3..5 buttons on which user can assign a macro as you wish.
If in the settings it will be possible to set the behavior of the button 
- hidden permanently (if not used),
- hidden (or inaccessible) during the printing process, 
- hidden (or inaccessible) during the bedleveling process,
the solution will be universal (f.e. see issue #156).

Someone will be able to assign BL-Touch sensor control commands to these buttons (lower / raise the probe), someone will reset the printer.
Personally, I really do not have enough buttons to send the M374 command to save the table created by the well-doned bed level adjustment in the controller's memory.

LMS0815, 2019-11-19:
I have the same issue. 
As a workaround, I am using OctoPrint "Custom Control Editor" to create those missing buttons.

jneilliii, 2020-01-19:
So i have made some headway into getting this feature implemented.  If you'd like to test it out for me that would be great.  You can install my latest development version by using the URL below in Plugin Manager.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.12.zip

[image]

[image]

EagleB3, 2020-01-21:
Yesterday I started testing the new version. Just in case, I’ll say that I installed the modified plugin by simply unpacking the files from the ZIP archive and replacing the existing files in the plugins directory with Raspberry.

So far so good! Excellent functionality, very convenient to use! You even made pictograms on the buttons - **EXCELLENT**!
I have not found any errors in the work, I will continue testing.

And, I think it will be very useful to add the ability **to assign Hint** for this buttons (pop-up 
tooltip) to the interface. This can probably be done something like this:
[image]
Probably, the maximum number of characters in such a hint should be limited by some reasonable number (100? 255?).

It seems to me that adding such an opportunity is not difficult, and with it the functionality will become absolutely universal.

Thank you for your work!

jneilliii, 2020-01-21:
The tooltip is set on the buttons and should be popping up the gcode commands, iis that not enough? 

My TabOrder plugin allows for setting the icons on the tabs.

EagleB3, 2020-01-21:
The fact that you did pop-up G-codes is very good. It is wonderful! Let them pop up if the user has not set their own Hint for the button.
You have made excellent universal buttons that can send not only the "M119" command to the terminal, but also a sequence of commands. It’s not convenient to parse a long sequence, Hint is more convenient
One person can program the button, and another can use it, still less familiar with G-codes.
It will be possible to make small-sized buttons, with only one icon, and at the same time having good information content. The script from the screenshot works great on the button you made. But it will be much better if not the full G-Code pops up, but the phrase "**Define Z0 for print begin**"
By the way, the script in the popup field did not fit all. The last two characters did not fit, they are not visible.
[image]
Therefore, I believe that such a hint is advisable.

This is a higher level of ergonomics. Higher level of service for the user. This is a programming culture. We must strive for this.

But of course, this is just IMHO. You certainly have already done a very good job. Thanks!

jneilliii, 2020-01-21:
Ok, you make a good point, but this is not my job, I make very very little on the tips I get via PayPal relative to the time I put into making changes and implementing feature  requests. 

BTW, you should be able to enter the commands with returns instead of the | symbol since I converted it to a text area.

jneilliii, 2020-01-22:
Slightly changed the layout. Made a little more sense putting the tooltip option on the same line as the icon, label options as they're all related to the button itself.  That allows for a larger area for the command field. I've just pushed an update to the same branch that incorporates the option.  

[image]

[image]

[image]

**Warning on installing this update**.  Either delete all your custom commands prior to upgrading or manually modify your config.yaml adding a `tooltip` field to each custom command like below to either what you want your tooltip to be, blank,  or `null`.

```
plugins:
  bedlevelvisualizer:
    commands:
    - command: '@BEDLEVELVISUALIZER

        M117 Label 1 Clicked'
      enabled_while_graphing: false
      enabled_while_printing: true
      icon: fas fa-archive
      label: Label 1
      tooltip: Update Mesh Data
    - command: M117 Label 2 Clicked
      enabled_while_graphing: true
      enabled_while_printing: false
      icon: fas fa-expand-arrows-alt
      label: Label 2
      tooltip: 
    - command: M117 Label 3 Clicked
      enabled_while_graphing: true
      enabled_while_printing: false
      icon: far fa-angry
      label: Label 3
      tooltip: null
```

EagleB3, 2020-01-24:
Before installation, I deleted all previously made additional buttons. I was somewhat puzzled: I saw that after the update in the "config.yaml" file there are new lines of "tooltip:" for the newly created buttons after saving the results, but they do not appear on the screen and restarting the octoprint and raspberry server did not change anything. New fields were displayed only after deleting the browsing history (Ctrl + F5). I did not immediately think of it...

I have been using the new version of the plugin intensively for the third day - **while there are no complaints about its functionality. I did not find any errors or flaws. Everything works great. Fine!**

I believe that you have added a very convenient tool! It seems to me that a panel with such universal buttons would be very popular among users on the "Terminal" tab.

And I allow myself to give two more small tips on the interface:
1) I hope you do not mistake me for a maniac, but add a tooltip to the "Update" button itself. It will be good if this tooltip makes the name of the corresponding field in the plugin settings intuitive.

2) Slightly increase the "Update" button itself - this is the most important button, but it is quite invisible (despite the fact that to the left and to the right of it there is a lot of free space, which has nothing to occupy). And the amount of text on the button can be increased.

Thank you so much for the work done!

[image]

jneilliii, 2020-01-24:
Thanks for the feedback. I'll include those changes during my next round of edits. 

> It seems to me that a panel with such universal buttons would be very popular among users on the "Terminal" tab.

This is what the custom control plugin does.

https://plugins.octoprint.org/plugins/customControl/

EagleB3, 2020-01-24:
You are welcome! Glad to help in a good deed!

Custom controls, in my opinion, have three important advantages:
1) It allows you to create buttons with a request to confirm the launch (**this is very important**)!
2) It allows you to remove groups of buttons in the spoilers;
3) It allows you to run external macro files (* .gco)

And yet I think your buttons are more convenient. Yours:
1) There is a tooltip. This is VERY, VERY necessary for rarely performed operations (such as setting the auto-level sensor itself, which is performed every three months or less);
2) You can make buttons miniature (an icon or a minimum of text, tooltip will not let you get confused)
3) You can enable / disable the buttons during printing / while viewing.

Most importantly: "Custom controls" works on the "Controls" tab, and **not on the "Terminal" tab**.
Until the custom buttons appear on the “Terminal” tab, users will have to switch to the “Controls” tab (or BedVizualizer :)), press the button there, and then switch back to the “Terminal” tab.

So you should consider the plug-in of universal buttons for the “Terminal” bookmark!

jneilliii, 2020-01-24:
> So you should consider the plug-in of universal buttons for the “Terminal” bookmark!

Oh right, they do add to the controls tab. There was another plugin I vaguely remember that would move those to the terminal tab, maybe in the community forum.

jneilliii, 2020-01-25:
I thought I saw one for the terminal tab. Buttons don't have icons, but it does give you the buttons. 

https://plugins.octoprint.org/plugins/TerminalCommands/

EagleB3, 2020-01-25:
Many thanks! I will definitely use it!

jneilliii, 2020-01-25:
Just pushed updates to the same branch for additional testing.  Incorporates your suggestions plus a couple of more UI and back-end server changes.  If you don't get prompted to reload after installing make sure to do the force refresh procedure you mentioned before. Sometimes when there isn't a version change on a plugin it won't flag OctoPrint to reload cache on its own.

jneilliii, 2020-01-26:
@EagleB3, I just uploaded a test plugin for terminal tab.  You can find it over here.

https://github.com/jneilliii/OctoPrint-TerminalCommandsExtended

EagleB3, 2020-01-28:
Everything works correctly in the BedVisualizer plugin (including the installation process from a ZIP file). After installation, it required a reboot, the interface was updated, the previously configured commands remained in place - no one disappeared, no one skewed.
I have been operating for two days, no errors were found.

Great functionality!

Well, a tiny remark in appearance:
[image]

EagleB3, 2020-01-28:
The **TerminalCommandsExtended** plugin is installed from a ZIP-file. There were no problems with the installation.
A reboot after installation was requested and completed.
I have been using this plugin for two days too. No errors appeared, everything works correctly.

And no comments on the interface.
And even without wishes for changing the interface - everything is perfect! :)

Great job. Thanks!

EagleB3, 2020-01-28:
Forgot to write right away. The "Cancel" button in the **BedVisualizer** plugin on my printer is problematic to test; immediately after starting a relatively short set of G-commands, followed by "G32" and displaying the results. But I will try. If errors are noticed - I will write.

jneilliii, 2020-02-02:
> The "Cancel" button in the BedVisualizer plugin on my printer is problematic to test

I have the same issue with testing using the virtual printer.  Would be good to know what firmware you are using and the gcode commands being used in the plugin's settings though.

EagleB3, 2020-02-03:
Hello! First, about the work of plugins.
Today I noticed the strange behavior of the "Terminal Commands Extended" plugin (using Mozilla Firefox 72.0.2 64bit).
There is no reaction to the "Add +" button in the interface.
But if you press it (say, three times), then save the work results (by clicking the Save button) and refresh the page (Ctrl + F5), then three empty buttons will appear.
These buttons can be edited, and after that they will be displayed on the Terminal tab.
With the removal of empty (not edited buttons) the interface behavior is similar. Clicking on the field (x) in the description of the button does not delete it; it will disappear only after the Save button is saved and the interface is updated by F5.
But with the edited buttons (which have contents in the Label and Command fields), this does not happen - after clicking on the field (x) in the description of the button, such a button disappears immediately.

The interface of the **BedLevel Visualizer** plugin regarding adding / removing buttons **works correctly, without complaints**.

Yesterday I performed “Update-Upgrade-Clean” on my RPi3, I noticed the effect today. But, as they say in detective novels, "after that - does not mean that as a result of this." Perhaps the reason is this, but... I don’t know. Not sure.
[image]

EagleB3, 2020-02-03:
Regarding the "**Cancel**" button:
I am using MKS SBase v.1.3 controller and Smoothieware v.1 firmware (for LPC1769 controllers)
Here is the code that is launched by clicking the "Update Mesh Now" button:
`G91`
`G0 Z+5`
`M400`
`G90`
`G28`
`G0 X78 Y101 F2000`
`M400`
`G30 Z0`
`G0 Z5 F1200`
`M400`
`G32`
`@BELEVELVISUALIZER`
Before executing the **G32**, the firmware I modified myself releases the probe (with a servo, I use the BFP-Touch probe), and at the end it removes it.

The only command that can interrupt work in Smoothieware is **M112**.
But it does not immediately stop the printer, and even the execution of the Mesh update procedure started with the **G32** command will not be interrupted.
If necessary, I can write more about this.

More appropriate is the option that will turn the "**Cancel**" button into a signal change on one of the PIO Raspberry, and already this PIO can be connected by wire to the contact of the hardware "**Kill button**" of the controller.

jneilliii, 2020-02-03:
Thanks for this feedback, but unrelated to the buttons.  This is a similar issue posted in #182 and I'm looking into a solution for that.  The problem is that the cancel flag is not stopping the gcode processing that's already been sent to the printer, so it still collects the mesh.  Also, I've moved your Terminal Commands Extended issue to the appropriate repository, please respond there.

jneilliii, 2020-02-07:
I've made some UI improvements to the custom commands section and fixed up the cancel issue. For cancel related input please provide input over at #182.

bakman2, 2020-02-14:
Some brainstorming: 

If one could click a point on the chart and get the values into a form-field, which can be adjusted (and possibly realtime updated in the chart to see the change) and then assign a button to send an command to manually update the stored mesh with the new value with a M421 (to create a "perfect" mesh).

jneilliii, 2020-02-14:
So there is a button in the graph with arrows that you can click that will enable the click to move to that point functionality. You would just need to add a button to make the adjustments. Can't remember if I integrated the input options to the buttons yet like I did for terminal commands extended.

PR #184 (2020-03-15): 0.1.12
#### Added
- Rotation options to mesh visualization, community forum request.
- Custom debug logging for future troubleshooting
- Custom commands that will give additional buttons on tab, resolves #166 
- Cancel button to stop processing when visualization gets stuck, resolves #182 
- Larger precision number processing, resolves #177
- Themify support, will now color the background and axis based on tab colors, resolves #181 
- Marlin Mesh Bed Leveling support, thanks to #186 
- Marlin OpenScad Output support, thanks to #186 
- Option to show webcam while processing, resolves #191 

### [... cut]
closed 2020-03-15 by jneilliii