jneilliii/OctoPrint-BedLevelVisualizer #168: Write Mesh capability

jtkdaddy, 2019-11-20:
Any plans to allow overriding mesh data?  My EZ-ABL sensor is somewhat inaccurate at reading a few points on the bed (3 points out of 15) and it would be a great feature to be able to compensate for those.  Also, in addition to the existing update button, can you add a button to just read the current stored mesh data in firmware?

jneilliii, 2019-11-20:
The request for different buttons has already been asked for. In theory you can do that now (assuming recent version of marlin) using M420 command I believe. Of course there are caveats based on firmware. If you're using UBL you can click on the icon at the top to enter adjust mode and when you click on the point in the visualization it will take your nozzle to that position allowing to adjust the mesh manually using G29 P4 T.

jtkdaddy, 2019-11-20:
M420 doesn't update the visual representation of the bed though.  Also, using G29 to manually adjust is tedious.  It would be nice if clicking on the z value in the bed visualization would display an edit field for the user to manually override the value.

jneilliii, 2019-11-21:
Like I said, very firmware dependent and since the plugin is really just reading data from a report I feel it would be difficult to implement. The changes to enable an unlimited command button feature may be able to include an input option to prompt for values passed to a gcode command that you configure.

closed 2019-12-12 by stale[bot]