jneilliii/OctoPrint-BedLevelVisualizer #170: Settings for large printers?

Quintin9801, 2019-11-21:


jneilliii, 2019-11-21:
Need to see what's returned from the `G29 T` command in the terminal tab and verify you have your printer profile in OctoPrint's settings configured correctly dimensionally.

Quintin9801, 2019-11-25:
Here is my output for G29 T, my bed size is set to 610x610 instead of the full 660x660 for safety, if this is the issue it can surely be changed.

> Send: G29 T
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     (  0,610)                                                              (610,610)
Recv:         0       1       2       3       4       5       6       7       8       9
Recv:  9 | -0.297  -0.133  +0.105  +0.314  +0.525  +0.796  +1.066  +1.414  +1.794  +2.175
Recv:    |
Recv:  8 | +0.323  +0.490  +0.700  +0.877  +1.055  +1.302  +1.535  +1.849  +2.184  +2.518
Recv:    |
Recv:  7 | +0.811  +0.967  +1.160  +1.320  +1.461  +1.671  +1.860  +2.124  +2.412  +2.700
Recv:    |
Recv:  6 | +1.307  +1.416  +1.615  +1.749  +1.853  +2.020  +2.172  +2.380  +2.612  +2.844
Recv:    |
Recv:  5 | +1.742  +1.841  +2.004  +2.104  +2.166  +2.319  +2.418  +2.576  +2.759  +2.942
Recv:    |
Recv:  4 | +2.070  +2.120  +2.240  +2.302  +2.342  +2.449  +2.508  +2.624  +2.776  +2.928
Recv:    |
Recv:  3 | +2.401  +2.371  +2.454  +2.470  +2.438  +2.518  +2.540  +2.629  +2.748  +2.866
Recv:    |
Recv:  2 | +2.685  +2.617  +2.618  +2.553  +2.512  +2.543  +2.535  +2.574  +2.658  +2.742
Recv:    |
Recv:  1 | +2.913  +2.725  +2.673  +2.567  +2.470  +2.463  +2.413  +2.407  +2.448  +2.488
Recv:    |
Recv:  0 |[+3.141] +2.833  +2.728  +2.582  +2.470  +2.463  +2.413  +2.407  +2.448  +2.488
Recv:         0       1       2       3       4       5       6       7       8       9
Recv:     (  0,  0)

jneilliii, 2019-11-25:
Wow, that is a huge variation from point 9,0 to 0,0.  What is happening is your points are outside of the visible graphing range currently.  Does your bed have any form of manual leveling possibilities? If it does, then getting the bed more level in order to reduce the offset from the lowest point to the highest point would resolve this. If not the only thing I can think of is changing the z dimension passed to the graphing library to be larger than the +/-2mm offset from 0 that is configured for now. IMHO getting your bed more physically level will be better for you in the long run anyway.

Quintin9801, 2019-11-25:
There is a manual way of leveling the z between 3 different steppers, the "tilt calibration" as they call it is measuring the nozzle height at each stepper position. I will definitely do more testing to get that range down. I probably should have been looking at this myself in hindsight. Thank you so much for all you do in the OctoPring Community!

closed 2019-11-25 by Quintin9801