jneilliii/OctoPrint-BedLevelVisualizer #171: {question} - Is it possible to update the mesh level map at the beginning of every print?

kquinsland, 2019-12-15:
Hi there. First, thanks for developing this plugin. I am so grateful for the entire octo-pi ecosystem and the amazing community of plugins!

Perhaps this is obvious, but i was wondering if there's anything that i can put in the pre-print gcode that will get the visualizer to build a new plot after every measurement procedure.

I see from the `readme` that you have the option to change what gcode is sent to the printer to get the data... but it's not clear to me if i can just put similar gcode into my pre-print settings or if clicking the `Update` button is required is `@BEDLEVELVISUALIZER` the magic word that tells the plugin that updated data is about to be dumped to the console?

Thanks for your time :)

jneilliii, 2019-12-15:
Definitely possible. If you use the same flag in your start gcode settings the mesh will update.

closed 2019-12-15 by jneilliii

kquinsland, 2019-12-15:
So to clarify / document:

- the only way to generate a new visualization after the mesh level before each print is to add a bit of gcode that looks like this (for a mk3):

```
M155 S0 ; stop reporting temperatures so mesh level data will not be corrupted
G80 ; mesh bed leveling
@BEDLEVELVISUALIZER ; clue the viz plugin that we want it to listen...
G81 ; check mesh leveling results. This stores the leveling results for use with the Octoprint Prusa mesh bed leveling plugin.
M155 S0 ; begin reporting temperatures every 3 seconds again
```

And this magic `@BEDLEVELVISUALIZER` string is only going to trigger the plugin if the gcode is run from octoprint, correct? Or will i be able to use the printer's native print from SD card function to trigger this as well?

jneilliii, 2019-12-15:
You are correct, only files sent from local storage will trigger the plugin. The last M155 should be with a S3 for 3 second temperature reporting. S0 iis for disabling.

kquinsland, 2019-12-15:
> You are correct, only files sent from local storage will trigger the plugin. 

Ok thanks!

> The last M155 should be with a S3 for 3 second temperature reporting. S0 iis for disabling.

Good catch.  Turns out, the mk3 does not support the `m155` command so it's moot  :/